robotemi/sdk #194: IOT devices using Hey Temi

arpit-temi, 2021-04-30:
Hi 
Can I able to turn on/off **IOT** devices like smart bulb using Temi SDK? Can I able to do it without programming using "**Hey Temi**"?  

Thank You in Advance

zjn0505, 2021-05-02:
The easiest way to control IOT devices without programming on temi should be using Alexa

arpit-temi, 2021-05-02:
Is it possible to turn on/off bulbs using `Hey Temi`?

zjn0505, 2021-05-03:
There is no out-of-box IOT devices control with "Hey Temi".

In my view, the most important thing for IOT devices is how they can be controlled by developers. i.e. without "Hey temi", is that possible to control your devices by code?

If yes, then you are just few steps away to control them by "Hey temi".

arpit-temi, 2021-05-03:
Thank You for you valuable feedback. @zjn0505 If I understand this right. If I add smart bulb in Alexa and if Alexa provide that accessibility to developer then we can use that smart bulb using **Temi ASR** functionality. Right? 

Looking forward to your thoughts on this.

zjn0505, 2021-05-06:
If you have the smart bulb connected to Alexa account, and you can talk to Amazon echo to control the bulk.
Then just link to your Amazon account on temi and say to it "Alexa, turn on my light".
It should work without any development.
You can check this video for how to set up Alexa on temi. https://www.robotemi.com/tv/setting-up-alexa/

closed 2021-07-22 by lcgao