robotemi/sdk #195: Import Error came on Building Apk

Robotrator, 2021-05-05:
The project was working fine, but after I integrated with another project then some errors starts coming. It stops installing on Temi, and gives Error while Gradle Buid.

`com.robotemi.sdk.ISdkServiceCallback: couldn't find import for class com.robotemi.sdk.ISdkServiceCallback
com.robotemi.sdk.TtsRequest: couldn't find import for class com.robotemi.sdk.TtsRequest
com.robotemi.sdk.DisplayListRequest: couldn't find import for class com.robotemi.sdk.DisplayListRequest
com.robotemi.sdk.activitystream.ActivityStreamObject: couldn't find import for class com.robotemi.sdk.activitystream.ActivityStreamObject
` and so on ...

Robotrator, 2021-05-05:
After fixing the above issue, the below problem accurs.

> Task :temi:compileDebugKotlin FAILED
```
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (17, 40): Unresolved reference: ActivityStreamObject
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (18, 40): Unresolved reference: ActivityStreamPublishMessage
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (19, 40): Unresolved reference: ActivityStreamUtils
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (33, 45): Unresolved reference: OnCurrentPositionChangedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (34, 45): Unresolved reference: OnDistanceToLocationChangedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (35, 42): Unresolved reference: Position
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (36, 42): Unresolved reference: SafetyLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (37, 42): Unresolved reference: SpeedLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (38, 38): Unresolved reference: AlertNotification
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (39, 38): Unresolved reference: NormalNotification
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (40, 38): Unresolved reference: NotificationCallback
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (41, 36): Unresolved reference: OnRequestPermissionResultListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (42, 36): Unresolved reference: Permission
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (45, 38): Unresolved reference: CallState
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (96, 63): Unresolved reference: OnUsersUpdatedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (114, 29): Unresolved reference: OnRequestPermissionResultListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (117, 29): Unresolved reference: OnDistanceToLocationChangedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (120, 29): Unresolved reference: OnCurrentPositionChangedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (253, 57): Unresolved reference: Position
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (301, 61): Unresolved reference: CallState
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (342, 69): Unresolved reference: NotificationCallback
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (385, 25): Unresolved reference: Permission
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (398, 55): Unresolved reference: ActivityStreamPublishMessage
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (801, 32): Unresolved reference: Position
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (803, 25): Cannot access class 'com.robotemi.sdk.navigation.model.Position'. Check your module classpath for missing or conflicting dependencies
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (812, 27): Unresolved reference: SafetyLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (816, 24): Unresolved reference: SafetyLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (817, 53): Unresolved reference: SafetyLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (822, 20): Unresolved reference: SafetyLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (835, 20): Unresolved reference: SpeedLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (839, 24): Unresolved reference: SpeedLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (840, 46): Unresolved reference: SpeedLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (845, 20): Unresolved reference: SpeedLevel
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (876, 58): Unresolved reference: OnDistanceToLocationChangedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (881, 61): Unresolved reference: OnDistanceToLocationChangedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (886, 55): Unresolved reference: OnCurrentPositionChangedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (891, 58): Unresolved reference: OnCurrentPositionChangedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (1139, 45): Unresolved reference: OnUsersUpdatedListener
e: D:\OfficeProjects\TemiWayFinder\temi\src\main\java\com\robotemi\sdk\Robot.kt: (1149, 48): Unresolved reference: OnUsersUpdatedListener
[... cut ...]
> Task :app:mergeExtDexDebug
> Task :app:processDebugManifestForPackage

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':temi:compileDebugKotlin'.
> Compilation error. See log for more details

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 39s
31 actionable tasks: 10 executed, 21 up-to-date

lcgao, 2021-05-06:
Please check if both of these two projects are dependent on temi SDK in build.gradle, if so, exclude the dependency in one of them.

closed 2021-07-22 by lcgao