robotemi/sdk #197: Temi OS Version Upgrade Issue

PepperClass, 2021-05-25:
Hi 

After I tried to upgrade Temi OS Version to 
Launcher OS: 14654
Robot OS: 124.35

And my app using SDK version:
implementation 'com.robotemi:sdk:0.10.53'

My App is Crashed, errors list below:

E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.perobot.temibizsuite, PID: 9285
    java.lang.NullPointerException: Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.checkNotNullParameter, parameter packageName
        at android.os.Parcel.readException(Parcel.java:1626)
        at android.os.Parcel.readException(Parcel.java:1573)
        at com.robotemi.sdk.ISdkService$Stub$Proxy.toggleWakeup()
        at com.robotemi.sdk.Robot.toggleWakeup()
        at com.perobot.temibizsuite.MainActivity.onRobotReady(MainActivity.java:736)
        at com.robotemi.sdk.Robot.setSdkService()
        at com.robotemi.sdk.TemiSdkServiceConnection$1.onServiceConnected()
        at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:1223)
        at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:1240)
        at android.os.Handler.handleCallback(Handler.java:743)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:5417)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:772)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:662)

    @Override
    public void onRobotReady(boolean b) {
        if (b){
            robot = Robot.getInstance();
            //robot.hideTopBar();
            robot.toggleWakeup(true);       // Error (line:736)
            robot.toggleNavigationBillboard(false); 
            refreshTemiUi();
        }
    }

Then when I try to upgrade SDK version to :
implementation 'com.robotemi:sdk:0.10.74'

The problem is solved, 
I don't know if I have to upgrade all my older Apps(already running on customer's side Temi Robot) to prevent App crash?

Thank you for your help!

lcgao, 2021-05-26:
Hi, to upgrade your App and using the latest temi SDK(0.10.77) for Version 124 is the most efficient way. Actually using function [`toggleWakeup()`](https://github.com/robotemi/sdk/wiki/Utils#toggleWakeup) in version prior to 0.10.70 of temi SDK in versions after 13646 of Launcher is ineffective because of the [permission mechanism](https://github.com/robotemi/sdk/wiki/permission).

Sorry for the inconvenient.

PepperClass, 2021-05-31:
Hi , thank you for the quickly reply that help me to figure out what happened and what to do next!

closed 2021-05-31 by PepperClass