robotemi/sdk #202: Get Image of sequences

kilmhd, 2021-06-08:
Hy,

In the last release, you push a change on "Add parameters image key and description for Seqeunce".
I would like more information about retrieving images to then display them in drawable format.

I see this method: getSignedUrlByMediaKey()
But it's slightly unclear about the use for me.

Do you help me with an example or more information ?
Thank's

zjn0505, 2021-06-12:
Hi, sequence data will include image key for the sequence cover image.
It is not in the format of URL, but as a mediaKey, then pass it in a list to the API `getSignedUrlByMediaKey()`, then it will take query network storage and return you a temporary media URL. The URL will be valid in next few minutes, within this period, you shall be able to use it as regular imageUrl, to download it and convert it into drawable.

Example in the SDK sample:
https://github.com/robotemi/sdk/blob/72db2e9b9ba1e3ee60f04215c39332f23e9f4b05/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L1159-L1182

kilmhd, 2021-06-14:
Thanks for your answer, I will test that. :)

closed 2021-07-22 by lcgao