robotemi/sdk #212: MapDataModel.getMapImage() returning mostly empty MapImage class (Java)

DoubleWB, 2021-07-07:
Hello,

I am trying to load the map data into a bitmap, and send that bitmap as an image to a server. Several months ago (I don't exactly recall when) this was working. After upgrading to version 10.77 of the TemiSDK, the same code is failing due to the ```MapImage``` object being returned by ```getMapImage()``` being mostly empty. ```getData()``` returns null, ```getRows()``` returns 0, and ```getCols()``` returns 0, but ```getTypeID()``` returns valid data.

I have confirmed that the ```MapDataModel``` object I am calling ```getMapImage()``` on is mostly not broken, given that ```getMapInfo()``` call returns valid data. I've confirmed this data is valid by loading the Temi SDK Sample App on my temi, and comparing my output to the output of the ```MapActivity``` in that app. Everything is the same except for the ```MapImage``` containing no data in my app, but the Temi SDK Sample App doesn't have any issue displaying the ```MapImage```. 

I have tried to copy the code from the SDK Sample ```MapActivity``` as closely as I can - the only difference is that it does not have any UI code, and just stops after loading the ```MapImage``` into a bitmap. The only difference that I am aware of is that my code is written in Java, whereas the ```MapActivity``` in the sample app is written in kotlin. Could there be some functional difference between the kotlin and Java versions of these objects?

Update: A coworker of mine was able to get the ```MapImage``` data to populate by forcing the program to hang on the ```getMapImage()``` line for a very long time with the android debugger. Is it also possible that this call returns early when it should be blocking?

DoubleWB, 2021-07-07:
We've discovered this was caused by not matching the proguard rules of the Temi Sample App. If you are using minification or shrinking of any kind, you must have the line ```-keep class com.robotemi.sdk.** { *; }``` included in your ```proguard-rules.pro``` file in order for these sdk calls to function properly.

closed 2021-07-07 by DoubleWB

zjn0505, 2021-07-08:
> the MapImage object being returned by getMapImage() being mostly empty. getData() returns null, getRows() returns 0, and getCols() returns 0, but getTypeID() returns valid data

Also it might related to the size of the map.
We are aware of this issue that big map will take long to get the map data.
There is a regression on the 124 version, that big map will fail on this place.
We will append a fix in 125 version, and will refactor this to provide a better and faster map data in 126 version.