robotemi/sdk #214: uninstalling

VantieghemMarijn, 2021-07-14:
Hello I am a backend developper that wants to develop more for Temi.
I have installed an MQTT app for Temi that I want to further develop. I installed the app with the apk and that works fine.
When I want to update the app with a new apk it says that i cant install that app because it is already installed. So I wanted to delete that app with adb uninstall but I only get this : Failure [DELETE_FAILED_INTERNAL_ERROR].

Does anyone know what this could mean because it doesn't explain a lott.

Thanks in advance for replying.

leslieam, 2021-07-22:
To install that app, you can use adb install -r xxx.apk. -r option will help in such case.

What's the package name you're trying to uninstall. usually, it will show this error when removing a system integrated app which you can not remove it with adb uninstall.

VantieghemMarijn, 2021-07-22:
Hello

I have tried the option -r before with uninstalling and installing an app but didn't work still gor rhe same error. The package i am trying to uninstall is com.hapirobo.connect.

Kind regards
Marijn Vantieghem

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden

leslieam, 2021-07-23:
You can refer to [this](https://stackoverflow.com/questions/38892270/delete-failed-internal-error-error-while-installing-apk) for some help.

This is not a system app. You can uninstall it. I don't know why you got this.

If you still can not.
You can find the package name in these folders and remove it.

/data/app
/data/data
/data/dalvik-cache

Don't remove other packages

closed 2021-10-19 by lcgao