robotemi/sdk #216: Cross widget

kilmhd, 2021-07-19:
Hi, 

I put an ISSUE on the "temi-welcomingmode-b2b" GitHub on this subject.
I would like to know how to get this feature?  I would like to display a cross to return to my kiosk application, when the user is on another application, can you refer me?

Thanks

zjn0505, 2021-07-20:
I suppose you see a cross over other pages because its main activity is transparent.
You are see through the main activity of welcome b2b.
https://github.com/ROBOTEAM-HOME/temi-welcomingmode-b2b/blob/master/app/src/main/res/values/styles.xml#L8-L11

If you app is the chosen kiosk app, then clicking on home will start your app.

kilmhd, 2021-07-22:
Thanks for your answer. I will try this and launch another application in the background to always have the kiosk app on the first ground.

closed 2021-10-19 by lcgao