robotemi/sdk #218: Google Assistant on Temi?

panititry, 2021-07-29:
I have an app which used google assistance for TTS and I want to integrate it to Temi. I wonder does Temi support for using google assistant ?

zjn0505, 2021-07-29:
Please provide more details on how you integrate your app with Google Assistant.

For example, is your app consuming Google speech to text service via their cloud service. https://cloud.google.com/speech-to-text/docs/apis

Or you need to install Google Assistant app on the tablet, and having your app talk to it.

panititry, 2021-07-29:
On the second, Google Assistant which is on a tablet .

zjn0505, 2021-07-29:
This one I don't have experience on it
If there is a robot with you, you can try it yourself. Or if you can provide more technical details on the integration, we can check it on our side as well.

DanLechuga, 2022-01-24:
If you use the google assistant to carry out the actions, you have a response time of 5 to 10 seconds in which it responds for the activity results, depending on the speed of the internet, and the size of the application.
instead if you use temi's natural activator.
'Hey Temi', your time is reduced to 3 seconds.
I know because I develop on one. It doesn't take long for a new version of temi to come out with a smaller 360 sensor