robotemi/sdk #220: is it possible to change the wakeup word?

ashwindevadiga, 2021-08-29:
Need to change the wakeup word 'Hey Temi' to something else.

zjn0505, 2021-08-29:
At this moment, it is much more than a simple call of `setWakeupWord("Hello, robot!")`
So currently it cannot be provided as an SDK interface.

ashwindevadiga, 2021-08-30:
Ok. Is it at least possible to add additional wake-up words?

zjn0505, 2021-08-30:
In current software version this is not supported yet. We will update in the release note once it is feasible for end user / developers to do that.

rickyhello57, 2022-01-03:
Is this now available on release 127 ?

lcgao, 2022-01-07:
Please open a ticket on [Zendesk](https://robotemi.zendesk.com) for changing wakeup word request.

DanLechuga, 2022-03-05:
@rickyhello57 And if you configure the nlu, so that when you say the word, it activates, activates the voice assistant