robotemi/sdk #224: MapId does not match the id of getMapList.

kang0252, 2021-09-13:
Hello

MapDataMode.mapId refers to the current map ID of Robot, which does not have the same ID in the list of getMapList().

Is there a way to find a map that matches the id value of the getMapList?

kang0252, 2021-09-14:
Thank you for your quick response.

I try to manage and implement location information and map information in units of map ID.

When Temi Center sends the backed up map to the robot again, is there an API that can know the information of the backup MapId?

kang0252, 2021-09-14:
I confirmed that MapDataModel.mapId generates a new ID every time a map is rescanned. However, in getMapList(), all lists are returned.

Please check.

kang0252, 2021-09-14:
I think adding mapId to the MapModel will make it simple.

What do you think about it?

lcgao, 2021-09-14:
That's correct, the mapId of MapDataModel will change if current map changed. Currently there is no method to get the mapId from backed up map, developers can only get current map's mapId, we will think about adding the mapId to backed up map list or somewhere.

kang0252, 2021-09-14:
Thank you.
I'll wait for the update. Have a great day.