robotemi/sdk #225: Sequence tools

kilmhd, 2021-09-15:
Hi, 

I would like to know if is possible to get a listener to know what is the current sequence run.
In the SDK there is a listener for the status, but his not permitted to know the current sequence.

There is another possibility to get these pieces of information? Or this listener is for a future update?

thank's

zjn0505, 2021-09-15:
Would you want get info the content of ongoing sequence?

kilmhd, 2021-09-16:
I would like get the name or ID of the current sequence, to apply a specific action. In the idea, the listener return the name of the sequence and if this sequence name is "Welcome", I take specific action after the end of this sequence.