robotemi/sdk #226: Invoking an action once the robot has arrived at a location

LeoFau, 2021-09-23:
Hallo which method do I need to use, if I want to make sure the robot has arrived at a location?
Let's say I am sending him to the kitchen and I want him to say something specific when he arrived.
I am using the goTo-method to send him there. How do I make sure, the code "waits" until he gets there?

Thank you in advance.

lcgao, 2021-09-24:
Hello, use this listener to listen to the status of *go to prograss*: https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener.

Sample code:
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L67
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L120
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L149
https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L726-L737

closed 2021-09-24 by lcgao

LeoFau, 2021-09-24:
I have tried to use this method, but the robot still speaks before reaching the location.

I tried to Override the method to do, what I needed in certain locations:
```
@Override
    public void onGoToLocationStatusChanged(@NotNull String s, @NotNull String s1, int i, @NotNull String s2) {
            if(userPosition.equals("kueche")){
                recipeRequest();
            }else{
                callRequest();
            }
    }
```

I also added the lines of code in the onStart and onStop and in the class. When trying to use the method I added it into my code like this:
```
robot.goTo(userPosition);       
onGoToLocationStatusChanged(userPosition, "complete", 1, "update");
```

This might be very wrong, but I find understanding the documentation very hard, so I would really appreciate the help.

lcgao, 2021-09-26:
This is an observer pattern, you don't need to invoke onGoToLocationStatusChanged() by yourself and SDK will invoke it when the status of go to location changes. So you just need to observe this listener(add listener at onStart()). For the meaning of onGoToLocationStatusChanged(), please check the documentation [here](https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener).

For example, if you want your temi say "Hello" when robot arrives location "kueche", you should add codes to the method as follow:

``` java
@Override
public void onGoToLocationStatusChanged(@NotNull String s, @NotNull String s1, int i, @NotNull String s2) {
    if ("kueche".equals(s) && "complete".equals(s1)) {
        robot.speak(TtsRequest.create("Hello", false));
    }
}
```