robotemi/sdk #238: Is there a way to programmatically have the temi listen STT?

mbncod, 2021-10-29:
Quick context: Building a custom dialog agent application for the Temi. Wanted to see if there's a way to capture user speech input without using the wake word. For example, our application dialog asks the user a question via our own speech (not using the native temi TTS, which we have disabled) and right now we have to say "Hey Temi" to invoke the native STT prior to giving our conversation response. Is there a way to invoke the AsrListener without having to use the wake word?

lcgao, 2021-11-01:
Hi, please consider using [`Robot.getInstance().wakeup()`](https://github.com/robotemi/sdk/wiki/Speech#wakeup) to trigger the STT.

mbncod, 2021-11-04:
Hi, Ive tried implementing Robot.getInstance().wakeup(), with no success. I am triggering this after our native TTS is hit so that it can listen for the user's response but the Temi does nothing when thatt method is hit.

mbncod, 2021-11-04:
For example, here's an attempted implementation where onRobotReady, I want the Temi STT triggered:

```
@Override
public void onRobotReady(boolean isReady) {
    Log.d("onRobotReady(temi)", String.valueOf(isReady));
    if(isReady){
        Robot.getInstance().wakeup();
        Log.d("onRobotReady(temi)", "wakeup()");
    }
}
```
Also tried the following, which was the only implementation of this method I could find on github:
```
@Override
public void onRobotReady(boolean isReady) {
    Log.d("onRobotReady(temi)", String.valueOf(isReady));
    if(isReady){
        runOnUiThread(new Runnable() {
            @Override
            public void run() {
                Robot.getInstance().wakeup();
                Log.d("onRobotReady(temi)", "wakeup()");
            }
        });
    }
}
```

Log.d prints when robot is ready, but Temi STT does not trigger. I've tried this wakeup() method in several places no luck.  Any advice?

lcgao, 2021-11-05:
Hi, in order to make sure it works, I tested it several times, but there was no problem. Did you [override the default STT](https://github.com/robotemi/sdk/wiki/Speech#overrideAsr) or [override the conversation layer](https://github.com/robotemi/sdk/wiki/Speech#overrideConversationLayer)?

mbncod, 2021-11-05:
No but I did override the default TTS. I just tried removing that and still no luck.

saboutal, 2021-11-08:
I have the same probleme.

lcgao, 2021-11-10:
> No but I did override the default TTS. I just tried removing that and still no luck.

I mean overriding STT(Speech to Text) not TTS(Text to Speech), calling function `Robot.getInstance().wakeup()` (does not require any special permissions) has the same effect as voice wake-up("Hey, temi").

mbncod, 2022-02-17:
Btw, I have pulled the sample code from this repo and ran it on my Temi. The sample code is a set of button commands and a logger on the right side. I successfully ran the code and was able to use it, BUT replacing code for the "disable wakeup" button to instead run `Robot.getInstance().wakeup()` does nothing. Here is the code I replaced in the sample code:

    private fun disableWakeup() {
        if (requestPermissionIfNeeded(Permission.SETTINGS, REQUEST_CODE_NORMAL)) {
            return
        }
        Log.d("disableWakeup(?):", Robot.getInstance().isReady.toString());
        Robot.getInstance().wakeup();
    }
    
The log is printed, but the wakeup() method does not work on my Temi.

lcgao, 2022-02-18:
Hi, did you set your App as the Home Screen(Kiosk mode)?

mbncod, 2022-02-18:
Was not trying to disbable wakeup(), instead replaced the functionality on this function in the sample code to see if I could get wakeup working. Which it did not work.

I actually was FINALLY able to get the wakeup() function working in both my application and the sample code in this repo. I could no longer figure out what the issue was in the code and just tried a factory reset of the Temi itself. Once the Temi was factory reset the same code with no changes were able to run the wakeup() commands. Maybe my Temi itself is defective? I dont know, but unfortunate to have these issues when the robot itself is so expensive.

prashantsmp, 2023-08-11:
If you are activated Kiosk mode in your application, Robot.getInstance().wakeup() will not work.