robotemi/sdk #242: Disable tracking function

MiRobotic, 2021-11-13:
Hi

We have customize a app and one function is required to detect person and greeting. So the greet mode is on. We disable the setting tracking people function in setting. However, the robot still tracking people when it detect person.

How to disable tracking people but greet mode still on

Thanks

lcgao, 2021-11-15:
Hi, you can configure this on [temi Center](https://center.robotemi.com/):

[image]

MiRobotic, 2021-11-15:
> Hi, you can configure this on [temi Center](https://center.robotemi.com/):
> 
> [image]

Hi, I still need greet mode but i dont want to track a person. As everytime robot track a person, then it moving along. I already disable the button you mention above. However the robot still tracking person

Thanks

lcgao, 2021-11-19:
Shouldn't be like that, could you please shoot a video for it?

leslieam, 2021-11-19:
How did you trigger user tracking, with a touch on the main button on the head?
This will trigger following not user tracking.

User tracking is trigger in greet mode. You need to configure the greet mode and wait the robot enter greet mode and then walk toward it to be in front of the robot.

closed 2022-03-22 by leslieam