robotemi/sdk #243: Display the location name in all capital letter

MiRobotic, 2021-11-13:
Hi

We want the location name display in all capital letters. We already saved the location name in all capital letters. However, only first letter is capital letter, the rest is small letter. 

Any way we can change it to all capital letter?

Thanks

zjn0505, 2021-11-14:
Location name is saved lowercased, meaning you cannot save 2 locations with same name but different case, and the all-capital letters will be saved as lowercase.
Only in the robot UI and temi center UI, the locations are displayed with 1st letter uppercased.

If you use https://github.com/robotemi/sdk/wiki/Locations#getLocations, it shall return locations with lowercase.

MiRobotic, 2021-11-14:
> Location name is saved lowercased, meaning you cannot save 2 locations with same name but different case, and the all-capital letters will be saved as lowercase. Only in the robot UI and temi center [...]

So which mean unable to save locations name in all uppercased?  Thanks

lcgao, 2021-11-19:
Exactly.

closed 2021-11-19 by lcgao