robotemi/sdk #244: virtual wall status

MiRobotic, 2021-11-13:
Hi

can we get a status which robot has across the virtual wall. It seems we only able to get repositioning status

Thanks

lcgao, 2021-11-15:
Hi, first of all, all SDK APIs is for current robot so you can't access the status or informations of other robots.

MiRobotic, 2021-11-15:
> Hi, first of all, all SDK APIs is for current robot so you can't access the status or informations of other robots.

My bad, I mean how to capture my current robot which is outside the virtual wall

lcgao, 2021-11-19:
There is not any specific APIs to capture that. But you can get the current coordinate of the robot and get the coordinates of virtual walls by following APIs:

https://github.com/robotemi/sdk/wiki/Locations#onCurrentPositionChangedListener
https://github.com/robotemi/sdk/wiki/Locations#getMapData
https://github.com/robotemi/sdk/wiki/Locations#mapDataModel

So you can think about how to use these APIs to implement your requirement.

iamkhan001, 2021-12-09:
Hi @lcgao is there any way to get coordinate for location name?
I guess is robot is not moving to location accurately, then we are planing to match with current coordinates and location name coordinates