robotemi/sdk #245: robot stuck at calculating

Alex-Beh, 2021-11-15:
The robot is stuck at **CALCULATING**. I tried to use the call the stopmovement() API but it didn't solve the issue.....

Need help urgently.....

leslieam, 2021-11-15:
Hi Alex, sorry to hear that.

Would you please check whether you can stop with touch the main button(The big one on the head)?
Also can you please share with us the serial number and a video of the symptom?

Alex-Beh, 2021-11-15:
Hi, 

Can I have your email address? So I can send your the serial number in th email?

Unfortunately I have restarted the robot. Will take a video when it happens again.

Besides, we try to send the robot to certain location. However anyone who touch the robot screen will stop the robot immediately...... Is there a way to stop this feature?

leslieam, 2021-11-15:
You can share it here and I'll remove it when  I record this.

You can try to use these apis to avoid this

https://github.com/robotemi/sdk/wiki/Utils#toggleNavigationBillboard
https://github.com/robotemi/sdk/wiki/Utils#setHardButtonMode

Also please provide android logs if the stopmovement() doesn't work.

Can you please tell me which distributor did you buy our temi robot from ?

Alex-Beh, 2021-11-18:
My robot serial number is 00121195583.

Thanks for your suggestions. Can I keep the "Going to home base" billboard status but just disable the touch screen stop navigation feature?

We bought temi robots from Mi Robotics.

leslieam, 2021-11-18:
Is the calculating issue is still happening?

Alex-Beh, 2021-11-22:
It didn't happen again so far.

closed 2021-11-22 by Alex-Beh