robotemi/sdk #246: What does mapdata mean?

WoutDemeyere, 2021-11-17:
Hello

I'm a little confused when using the getMapData() function. To what do the OriginX, OriginY & resolution variable of mapInfo refer?
And how do the coördinates of the locations and virtual walls correlate to them?

Many thanks in advance!

lcgao, 2021-11-23:
Hi, please check #184.

WoutDemeyere, 2021-11-24:
Yes thank you, i saw that issue but i'm more confused on how the coördinates correlate to each other. 

For example my OriginX & Y = -31.85... , -20.30... and my locationX & Y = -0.005..., -0.019.... Am i supposed to add these coördinates together or how do i interpret them? And what does the location theta mean?

Thanks!

lcgao, 2021-11-30:
Hi, the position is relative to the coordinates of the Home base and it is a right-hand system. It means the X would be a postive value if the position is front of the home base and Y would be a postive value if the position is on the left of the home base. Such as location(x=1, y=2, theta=1) means this location is  1m on the x-axis and 2m on the y-axis, and the theta is counterclockwise angle relative to the home base(the unit is radians, range(-π, π)).