robotemi/sdk #248: setGoToSpeed not working

Alex-Beh, 2021-11-25:
Hello, I have tried to use the API setGoToSpeed() but it is not working.

```
  SpeedLevel newspeedlevel = SpeedLevel.SLOW;
  sRobot.setGoToSpeed(newspeedlevel);
  Log.i(TAG, "[setSpeed] setSpeed: "+newspeedlevel);
  Log.i(TAG, "[setSpeed] currentSpeed: "+sRobot.getGoToSpeed().toString());
```

Current printout is:
```
I/MAIN: [setSpeed] setSpeed: SLOW
I/MAIN: [setSpeed] currentSpeed: HIGH
```

Expected printout is:
```
I/MAIN: [setSpeed] setSpeed: SLOW
I/MAIN: [setSpeed] currentSpeed: SLOW
```

Alex-Beh, 2021-11-30:
@zjn0505 @lcgao @hermanShai friendly ping

lcgao, 2021-11-30:
Sorry for the delay, please check below things:
1. App has the Settings permission
2. As the selected kiosk App and as the home screen

closed 2021-12-29 by lcgao