robotemi/sdk #249: customize app auto closed back to temi screen

MiRobotic, 2021-11-25:
Hi

We have developed one customized app. Set it as launcher and default it at temi center

However once a while temi screen pop up. Our app program to auto start again in 1 min if it is closed. We want to find out the root cause which lead to temi screen pop up.  Temi screen only pop up once a while, and no logic we can test it out. And we check the log also unable to see any error, no log at all to indicate what's the error and cause this issue

Anyone with similar issue? or how to solve it

Thanks

lcgao, 2021-11-30:
Hi, first of all please don't set your App as the launcher App of Android OS and please set it as the Home screen application in Settings > Home Screen > Applications instead. temi will back to temi's home screen in few seconds once no-interaction detected and it is the design, it means, your App will be launched once temi back to home screen if your App is set as the home screen application.

MiRobotic, 2021-11-30:
> Hi, first of all please don't set your App as the launcher App of Android OS and please set it as the Home screen application in Settings > Home Screen > Applications instead. temi will back to [...]

Hi,

temi home screen is displayed when we interact with robot also

thanks

MiRobotic, 2021-12-04:
@lcgao any further comment. thanks

lcgao, 2021-12-06:
Hi, if possible please shoot a video for us to check.

MiRobotic, 2021-12-06:
> Hi, if possible please shoot a video for us to check.

Hi 

Can I send you the link privately?

closed 2021-12-06 by MiRobotic

lcgao, 2021-12-06:
Yes, you can create a private Zendesk ticket for this issue: https://robotemi.zendesk.com.