robotemi/sdk #251: Robot lift alert

MiRobotic, 2021-12-04:
Hi

We has developed a program when the robot lift, there is alert msg sound out based on below sdk 
override fun onRobotLifted(isLifted: Boolean, reason: String) {
}

https://github.com/robotemi/sdk/wiki/Movement#onRobotLiftedListener

It is working when temi lift and alert msg sound out. However, sometimes there is no robot lift and the robot just going to certain location then robot also trigger it is lifted. Any comment

Thanks

zjn0505, 2021-12-07:
Lift detection is relying on the sensors around the robot, and there might be false alerts.

In our app we are holding/debouncing every lift signal for in 3 seconds, we will release the signal that the robot is lifted or not lifted if no new signal is received after 3 seconds.

This might help to filter those false positive alerts.

iamkhan001, 2021-12-09:
Hi @zjn0505 how to check alert is false alert ?
we have checked  **isLifted** value in **OnRobotLiftedListener** , but even alert is fake its **true**.

zjn0505, 2021-12-10:
We cannot tell a false alert, as it is sensor signals, can be affected by many things, maybe a piece of black floor that robot passed by.
What we did in temi app is we will wait for 3 more seconds as mentioned above for every true or false callback. and take the latest debounced result.

iamkhan001, 2021-12-13:
Thanks @zjn0505 , so once we got alert for robot lifted, we will wait for 3 seconds and once got alert again then consider it.
is it correct ?

zjn0505, 2021-12-23:
It is like postDelayed a 3-seconds message to emit the lifted event, during this 3 seconds, and false event will cancel the message, any new true event will cancel the message and create a new postDelayed message. 

> https://reactivex.io/documentation/operators/debounce.html
> https://reactivex.io/documentation/operators/images/debounce.png
> 
> [image]