robotemi/sdk #254: What is meaning of YAL in Position?

iamkhan001, 2021-12-27:
Hi there,

We are using **goToPosition()** of temi for navigation. here we are saving Position using x, y, yal and tiltAnlge. 
so can anyone here tell me what is **Yal** described https://github.com/robotemi/sdk/wiki/Locations#position here ?

lcgao, 2021-12-29:
Hi, it means the yaw angle relative to Home base location, the unit is radian.

iamkhan001, 2021-12-29:
thanks @lcgao

MiRobotic, 2021-12-29:
@lcgao The robot facing direction sometimes incorrect. How we can use program to force robot facing correctly. We able to get X and Y. And there is one value pop up in YAL. How to define it? or how we can make use of this data?

Thanks

lcgao, 2021-12-29:
There is no field called 'YAL' in SDK, it should be the 'yaw' of [Position](https://github.com/robotemi/sdk/wiki/Locations#position).
We provide listener to [listen to current position](https://github.com/robotemi/sdk/wiki/Locations#onCurrentPositionChangedListener)(x, y, yaw) so you can try to calculate the angle difference from the correct yaw of the destination location, and then use [turnBy(int degrees)](https://github.com/robotemi/sdk/wiki/Movement#turnBy) to adjust the direction

closed 2022-10-12 by iamkhan001