robotemi/sdk #256: getMapData() returning null

Ryan-Lee-2000, 2021-12-29:
Hi,

`Robot.getInstance().getMapData()`

I'm trying to get the Map Data from the current map but I'm getting a null. I've already allowed the Map permission in the android manifest.

lcgao, 2021-12-29:
Hi, only declared in AndroidManifext.xml is not enough, need to request the Map permission by [requestPermissions()](https://github.com/robotemi/sdk/wiki/permission#requestPermissions) or manually grant the permission in Settings > Apps & Permissions > Permissions > Map.

If still get the null from getMapData(), please check make sure the SDK is ready(listen to the [OnRobotReadyListener](https://github.com/robotemi/sdk/blob/master/docs/sdk/com.robotemi.sdk.listeners/-on-robot-ready-listener/index.md)) before calling it.

Ryan-Lee-2000, 2021-12-29:
Great, using [requestPermissions()](https://github.com/robotemi/sdk/wiki/permission#requestPermissions) to request the map permission worked out. Thanks!

closed 2021-12-29 by Ryan-Lee-2000