robotemi/sdk #258: How to check detected obstruction is a person or object and Patrol mode

ph4il2o, 2022-01-05:
1. I want to know about obstruction is a person or object when Temi is going on location.
 How can I do ?
2. I want to use patrol or survey mode in Temi. How to use it ?

lcgao, 2022-01-07:
1. Listen to [OnGoToLocationStatusChangedListener](https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener) and get the obstacle info from the `description` of the callback function `onGoToLocationStatusChanged()`. [Turn on the dection mode](https://github.com/robotemi/sdk/wiki/Detection-&-Interaction#setDetectionModeOn) and listen to [OnDetectionStateChangedListener](https://github.com/robotemi/sdk/wiki/Detection-&-Interaction#onDetectionStateChangedListener) at the same time to know if a person is detected.

2. How to start partol: *temi's apps > Locations > Bottom button (...) > Partol*. Developers can also implement another patrol via  [OnGoToLocationStatusChangedListener](https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener) in their App.

ph4il2o, 2022-01-07:
In the document it says
*temi will start the Detection Mode session if Launcher in a state without user interaction and movement, and your skill can listen for the events from that via this feature and according it to do what you want temi to do.

Can I use it when temi is go to Location?

ph4il2o, 2022-01-07:
I use OnGotoLocationStatusChangeListener and OnDetectionStateChange but when Temi is going to location and I Stand in front of Temi but onDetection doesn't work

I already set kios mode.