robotemi/sdk #259: Problems with ADB connection temi v127

jiashengmatteo, 2022-01-14:
Hi, I've just updated Temi to Robox OS v127.26 and Launcher OS 16013.

I've noticed that ADB connection has been changed and now requires a password needs to be passed via curl to 5556 port.
The problem is that, no matter how fast I type the command, It won't unlock and gives me following results:

```
$ curl -k "https://192.168.*.*:5556/grantAuth?pwd=2wmtxqDa"
{
  "status": "Failed",
  "error": "Error Code 02"
}
```
or
```
$ curl -k "https://192.168.*.*:5556/grantAuth?pwd=eGzfNUj1"
{
  "status": "Failed",
  "error": "Incorrect or Expired Password Provided"
}
```
I think it changes password too fast to be unlocked

zjn0505, 2022-01-14:
The password refreshes only onResume of that page.

jiashengmatteo, 2022-01-14:
Ok, I understood why it didn't work.
I was both connected to LAN and Wi-Fi network which are both in the same network with robot dedicated Wi-Fi SSID but my computer ip address wasn't in the same ip class with robot's ip (for example 192.168.1.* and192.168.2.*).
I didn't expect that this could be a problem, anyway now it works. Thanks

closed 2022-01-14 by jiashengmatteo