robotemi/sdk #261: ADB connection requires Password

LagrouCedric, 2022-01-18:
While trying to establish a ADB connection with temi, there was a new screen pop up that requires us to make connection with a password. This is probably part of the new secure ADB, but now we can't make connection to Temi, all 3 options to connect ( curl, http, adb connect) give timed out or host not found errors.

Is there another way of making connection to Temi with ADB? or another way of uploading apps onto the robot?

lcgao, 2022-01-20:
Unfortunately there is not another way to connect ADB, please provide the SN and the screenshoot for the issue to us, thanks.

LagrouCedric, 2022-01-21:
We made connection using a 4G hotspot, i think the 5556 port is blocked on our DNS server and so we could not establish connection.

closed 2022-01-21 by LagrouCedric