robotemi/sdk #263: Get Sequence IDs

knesslau17, 2022-01-19:
I am new to programming with Temi and would like to know how to get the sequence ids in java.
I would like to use the getAllSequences() method but don't know how to create a list with the ids.

Can somebody help please?

lcgao, 2022-01-20:
If you use Kotlin you can use `val ids = Robot.getInstance().getAllSequences().map { it.id }` to convert the sequence list to a seqeunce id list. And if you use Java you need to iterate over the entire sequence list and create a new String list to hold the ids of each sequence from sequence list, the following is the example in Java.

``` java
List<SequenceModel> list = Robot.getInstance().getAllSequences();
List<String> ids = new ArrayList<>();
for (SequenceModel sequence : list) {
    ids.add(sequence.getId());
}
```

knesslau17, 2022-01-21:
Thanks for the help! Really appreciate it.

closed 2022-01-21 by lcgao