robotemi/sdk #265: Allow Temi to work offline/in a closed network

evilmtv, 2022-01-31:
1. On startup, Temi is unable to proceed past the connect to wifi screen unless it connects to a wifi that has internet access
2. After successfully connecting to a wifi that has internet access and then changing to a wifi that has no internet access, Temi will randomly show this screen and will be stuck here until it is manually rebooted

[image]

It also randomly goes back to the home page when a custom app is running (it is not closed just minimized)

3. #260 Adb does not work on a network without internet access

Hope these can be fixed!

leslieam, 2022-02-07:
Hi @evilmtv 
1. This is by design.
2. Can you please get us the log and a video of what is happening? Did you see any crash?
3. Yes. Now it needs internet. We'll try to fix it in the new release.

evilmtv, 2022-02-17:
Hi @leslieam , thank you for the reply.

1) Would allowing reboot into a network without internet be considered? At least after the first time setup (which shows the introduction video and linking with Temi mobile app) is complete.
My project requires Temi to be on a closed network without internet access in order to connect with other IOT devices, and I would like to be able to regularly restart Temi programmatically without any manual changing of the wifi.
At the current state if I wish to restart Temi or if it crashes, I'll have to
- Select an internet accessible wifi on the prompt
- Key in password
- Once I get to the home screen, go back to settings
- Change wifi back to the closed network
- Forget the internet accessible wifi (To prevent Temi from connecting to the wrong wifi when it moves around)
- Start frontend app and resume job

2) I encountered it again just yesterday.

Temi is already booted and running fine beforehand
After switching to a closed wifi (no internet), Temi OS (not my app) seems to run slower where settings page takes longer to register input.
This screen sometimes pop up when starting my app.
There is no crash so far with the screen stuck there for a few minutes.

May I know how can I retrieve the log?
Next time I'll leave it there for 10mins to see if it crashes or is stuck forever on that page.

leslieam, 2022-02-18:
Hi @evilmtv ,

1. Temi would be able to run under a network without internet but the booting time(the 90s or more) will be much longer for we need the internet for temi's services to work normally.  The voice service, NLP, video calls, and so on services will not work properly if there's no internet.
2. Can you get us some videos and logs so that we can check it out.

evilmtv, 2022-02-18:
Hi @leslieam 

1. Currently it cannot boot to home screen without a wifi that has internet.
If the only saved wifi does not have internet, Temi will prompt a screen to select another wifi and I can't skip this step.

2. Ok, I'll update again when I record it

evilmtv, 2022-03-03:
Hi @leslieam , for the first point will there be a patch for this issue?

Also for the second point I am unable to record as it happens rarely and there is no clear way to reproduce it.
It can happen anytime while on a network without internet and has been a frustrating experience.
May I know how can I access Temi logs? If adb is the only way then unfortunately I can't view it as ADB is not currently possible without internet access as mentioned in the first post.

michael-soh, 2022-03-03:
Hi @leslieam, I am experiencing the same issue as @evilmtv.  After switching from wifi that has internet access to wifi without internet access, Temi would show that screen mentioned earlier and would often crash/ not responding. When it crashes, a pop up will appear saying "Unfortunately, Temi has stopped". Is it possible for Temi to run without the internet smoothly? It is incredibly annoying and frustrating to have to restart Temi and give it the internet accessible network(to get rid of the prompt to select another network) every time this happens.

leslieam, 2022-03-08:
@evilmtv 
When it happens. Can you please connect to a network with the internet and get us the log.

@michael-soh 
Would you please get us the log for "Unfortunately, Temi has stopped"?

evilmtv, 2022-03-08:
Hi @leslieam , managed to get some logs during a crash.

Steps to recreate:
Start in an internet-enabled wifi

1. Open our app
2. Go to wifi settings page from top bar wifi icon
3. Change wifi to internet-disabled wifi
4. Open app list page
5. Return to our app
6. App is very slow and eventually crash within ~3 seconds

However it occurs at random. Had to repeat steps a few times.

Crashed near 14:58:35

```03-08 14:58:33.089   576   726 W ActivityManager: Scheduling restart of crashed service com.roboteam.teamy.usa/com.roboteam.teamy.sdk.SdkService in 1000ms
03-08 14:58:33.090   576   726 W ActivityManager: Force removing ActivityRecord{b834974 u0 com.roboteam.teamy.usa/com.roboteam.teamy.internalactivities.list.mvp.activities.TabContainerActivity t2835}: app died, no saved state
03-08 14:58:33.090   576   629 E ConnectivityService: RemoteException caught trying to send a callback msg for NetworkRequest [ id=3, legacyType=-1, [ Capabilities: INTERNET&NOT_RESTRICTED&TRUSTED&NOT_VPN] ]
03-08 14:58:33.099   227   635 W AudioFlinger: removeEffect_l() 0xb35d1008 cannot promote chain for effect 0xb8a85cc8```

[Logcat full log](https://github.com/robotemi/sdk/files/8203587/8marcrash2.txt)
[Filtered to Warning and up](https://github.com/robotemi/sdk/files/8203586/8marcrash2W.txt)
[Filtered to Error and up](https://github.com/robotemi/sdk/files/8203588/8marcrash2E.txt)

leslieam, 2022-03-09:
@evilmtv 
I found the issue in your log.
Would you please disable bluetooth and try again.
It is because of the ANR in com.roboteam.teamy.usa which is due too high CPU usage especially com.android.bluetooth: 75%

03-08 14:58:32.860   576  3943 I WindowStateAnimator: createSurface WindowStateAnimator{a9f577f }: mDrawState=DRAW_PENDING
03-08 14:58:32.862   576   592 E ActivityManager: ANR in com.roboteam.teamy.usa
03-08 14:58:32.862   576   592 E ActivityManager: PID: 699
03-08 14:58:32.862   576   592 E ActivityManager: Reason: Input dispatching timed out (Waiting to send non-key event because the touched window has not finished processing certain input events that were delivered to it over 500.0ms ago.  Wait queue length: 4.  Wait queue head age: 10441.7ms.)
03-08 14:58:32.862   576   592 E ActivityManager: Load: 3.24 / 2.54 / 1.57
03-08 14:58:32.862   576   592 E ActivityManager: CPU usage from 262424ms to 0ms ago:
03-08 14:58:32.862   576   592 E ActivityManager:   78% 1514/com.android.bluetooth: 75% user + 2.4% kernel / faults: 16350 minor 25 major
03-08 14:58:32.862   576   592 E ActivityManager:   16% 699/com.roboteam.teamy.usa: 12% user + 4% kernel / faults: 89852 minor 214 major

evilmtv, 2022-03-09:
Hi @leslieam , it happens when bluetooth is turned off too.
It just happened again on another Temi today and it has bluetooth turned off.

evilmtv, 2022-03-10:
Here is another crash log with bluetooth turned off. (Only Warnings and up)

```
03-10 10:53:01.230   232  6061 E MediaPlayerFactory: 2019 03 30 -------- getPlayerType()---------------- 234 !!!!
03-10 10:53:01.230   232  6061 E MediaPlayerFactory: filePath = /data/app/com.roboteam.teamy.usa-1/base.apk
03-10 10:53:01.231   232  6061 E MediaPlayerFactory: 2019 03 30 -------- getPlayerType()---------------- return STAGEFRIGHT_PLAYER !!!!
03-10 10:53:10.033   563   592 W BroadcastQueue: Timeout of broadcast BroadcastRecord{6d4e647 u-1 android.intent.action.TIME_TICK} - receiver=android.os.BinderProxy@12a4d0f, started 10001ms ago
03-10 10:53:10.033   563   592 W BroadcastQueue: Receiver during timeout: BroadcastFilter{20eb7a5 u0 ReceiverList{b17919c 698 com.roboteam.teamy.usa/1000/u0 remote:12a4d0f}}
03-10 10:53:12.418   563   592 E ActivityManager: ANR in com.roboteam.teamy.usa
03-10 10:53:12.418   563   592 E ActivityManager: PID: 698
03-10 10:53:12.418   563   592 E ActivityManager: Reason: Broadcast of Intent { act=android.intent.action.TIME_TICK flg=0x50000014 (has extras) }
03-10 10:53:12.418   563   592 E ActivityManager: Load: 1.35 / 1.47 / 1.41
03-10 10:53:12.418   563   592 E ActivityManager: CPU usage from 10032ms to 0ms ago:
03-10 10:53:12.418   563   592 E ActivityManager:   9.8% 3521/firedomed-agent: 5.7% user + 4% kernel / faults: 66 minor
03-10 10:53:12.418   563   592 E ActivityManager:   6.1% 698/com.roboteam.teamy.usa: 5.1% user + 0.9% kernel / faults: 286 minor
03-10 10:53:12.418   563   592 E ActivityManager:   1.2% 232/mediaserver: 1.1% user + 0% kernel / faults: 87 minor 1 major
03-10 10:53:12.418   563   592 E ActivityManager:   1.7% 563/system_server: 1.1% user + 0.5% kernel / faults: 69 minor 1 major
03-10 10:53:12.418   563   592 E ActivityManager:   1% 202/dhd_dpc: 0% user + 1% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   1% 2754/sg.gov.tech.temi_changi_museum_app: 0.5% user + 0.4% kernel / faults: 5 minor
03-10 10:53:12.418   563   592 E ActivityManager:   0.3% 34/kconsole: 0% user + 0.3% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.8% 160/irq/196-mma8452: 0% user + 0.8% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.4% 77/fb-vsync: 0% user + 0.4% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.4% 201/dhd_watchdog_th: 0% user + 0.4% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.4% 5295/kworker/u8:1: 0% user + 0.4% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.3% 75/fb-vsync: 0% user + 0.3% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.1% 186/logd: 0.1% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.2% 8/rcu_preempt: 0% user + 0.2% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.2% 79/cfinteractive: 0% user + 0.2% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.2% 229/netd: 0% user + 0.2% kernel / faults: 42 minor
03-10 10:53:12.418   563   592 E ActivityManager:   0.2% 1064/wpa_supplicant: 0% user + 0.2% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.1% 652/sdcard: 0% user + 0.1% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0.1% 2209/kworker/0:2H: 0% user + 0.1% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0% 3/ksoftirqd/0: 0% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0% 14/ksoftirqd/1: 0% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0% 19/ksoftirqd/2: 0% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0% 24/ksoftirqd/3: 0% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0% 152/mmcqd/0: 0% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0% 191/surfaceflinger: 0% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   0% 684/com.android.systemui: 0% user + 0% kernel / faults: 2 minor
03-10 10:53:12.418   563   592 E ActivityManager:   0% 1452/com.google.android.gms.persistent: 0% user + 0% kernel / faults: 4 minor
03-10 10:53:12.418   563   592 E ActivityManager:   0% 2579/com.google.android.youtube: 0% user + 0% kernel / faults: 3 minor
03-10 10:53:12.418   563   592 E ActivityManager:   0% 7300/kworker/0:2: 0% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager: 5.9% TOTAL: 3.6% user + 2.1% kernel + 0% iowait + 0% softirq
03-10 10:53:12.418   563   592 E ActivityManager: CPU usage from 1841ms to 2361ms later:
03-10 10:53:12.418   563   592 E ActivityManager:   98% 698/com.roboteam.teamy.usa: 21% user + 76% kernel / faults: 184 minor 1 major
03-10 10:53:12.418   563   592 E ActivityManager:     96% 703/Signal Catcher: 23% user + 73% kernel
03-10 10:53:12.418   563   592 E ActivityManager:     1.9% 2659/ahpl_main: 1.9% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager:     1.9% 2811/RxCachedThreadS: 1.9% user + 0% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   3.8% 563/system_server: 0% user + 3.8% kernel / faults: 2 minor
03-10 10:53:12.418   563   592 E ActivityManager:     3.8% 592/ActivityManager: 0% user + 3.8% kernel
03-10 10:53:12.418   563   592 E ActivityManager:   3.6% 3521/firedomed-agent: 0% user + 3.6% kernel / faults: 4 minor
03-10 10:53:12.418   563   592 E ActivityManager:     1.8% 3719/firedomed-agent: 0% user + 1.8% kernel
03-10 10:53:12.418   563   592 E ActivityManager: 26% TOTAL: 6.7% user + 19% kernel
[... cut ...]
03-10 10:53:12.696   232  6061 W APM::EffectDescriptor: unregisterEffect() unknown effect ID 25
03-10 10:53:12.707   563   588 W AppOps  : Finishing op nesting under-run: uid 1000 pkg com.roboteam.teamy.usa code 24 time=0 duration=0 nesting=0
03-10 10:53:12.711   563   590 W AppOps  : Finishing op nesting under-run: uid 1000 pkg com.roboteam.teamy.usa code 24 time=0 duration=0 nesting=0
03-10 10:53:12.714   563  1267 W ActivityManager: Scheduling restart of crashed service com.roboteam.teamy.usa/com.roboteam.teamy.sdk.SdkService in 1000ms
```

leslieam, 2022-03-10:
This is another ANR issue not related with bluetooth.
What was the robot doing?

adb shell bugreport > bugreport.txt

Can you get me this report when it happens?

evilmtv, 2022-03-10:
Ok will share the file when it happens again.
It can happen when docked and idle or when moving. No specific action triggers it that I noticed

leslieam, 2022-03-10:
From the log, the system is doing something heavy.
So I want the log to understand way.
Also you can get me the anr logs with these 2 commands.
```
adb shell ls -al /data/anr > ~/logs/health_monitor/anrls.log
adb pull /data/anr .
```

evilmtv, 2022-03-14:
[crash14mar.zip](https://github.com/robotemi/sdk/files/8242467/crash14mar.zip)

Included are:
1. full logcat
2. anrls.log
3. anr folder
4. bugreport.txt

What was the robot doing:
- I was in the settings page switching wifi when this happened
- My own app was closed before the crash
- Bluetooth is turned off
- Temi was docked at home base, not moving/navigating

leslieam, 2022-03-22:
This is the crash in the log. 
Does it happen often to you with this crash?

@lcgao Please check it and try to fix it.
```
--------- beginning of crash
03-14 14:46:16.399   696   696 E AndroidRuntime: FATAL EXCEPTION: main
03-14 14:46:16.399   696   696 E AndroidRuntime: Process: com.roboteam.teamy.usa, PID: 696
03-14 14:46:16.399   696   696 E AndroidRuntime: java.lang.NullPointerException
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at com.robotemi.ros.ui.opengl.visualisation.XYOrthographicCamera.toCameraFrame(XYOrthographicCamera.kt:206)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at com.robotemi.ros.ui.opengl.visualisation.XYOrthographicCamera.toFrame$ros_release(XYOrthographicCamera.kt:218)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at com.robotemi.ros.ui.opengl.visualisation.VisualizationView.onTouchEvent(VisualizationView.kt:171)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.View.dispatchTouchEvent(View.java:9303)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2197)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2197)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2197)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2197)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2197)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2197)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2197)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at com.android.internal.policy.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2403)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1737)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.app.Activity.dispatchTouchEvent(Activity.java:2786)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at androidx.appcompat.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:69)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at com.android.internal.policy.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2364)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.View.dispatchPointerEvent(View.java:9523)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4230)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4096)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3642)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3695)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3661)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:3787)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3669)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:3844)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3642)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3695)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3661)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3669)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3642)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:5922)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:5896)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:5857)
[... cut ...]
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:135)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5417)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:772)
03-14 14:46:16.399   696   696 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:662)
```

leslieam, 2022-03-22:
@evilmtv 
I see in the anr/traces.txt
The trigger of the anr is in this app not our temi apps.
```
----- pid 30228 at 2022-03-10 16:53:23 -----
Cmd line: sg.gov.tech.temi_changi_museum_app
```

And due to the class file has been proguarded. We don't know why it happened.
You may need to check with this app.

```
"main" prio=5 tid=1 Native
  | group="main" sCount=1 dsCount=0 obj=0x73b14000 self=0xb7de5bd8
  | sysTid=30228 nice=0 cgrp=default sched=0/0 handle=0xb6ff4b34
  | state=S schedstat=( 9168916920 3778773610 38409 ) utm=598 stm=318 core=2 HZ=100
  | stack=0xbe2b1000-0xbe2b3000 stackSize=8MB
  | held mutexes=
  native: #00 pc 00042d4c  /system/lib/libc.so (__ioctl+8)
  native: #01 pc 000497d1  /system/lib/libc.so (ioctl+14)
  native: #02 pc 0001f0f5  /system/lib/libbinder.so (_ZN7android14IPCThreadState14talkWithDriverEb+132)
  native: #03 pc 0001f733  /system/lib/libbinder.so (_ZN7android14IPCThreadState15waitForResponseEPNS_6ParcelEPi+38)
  native: #04 pc 0001f8e9  /system/lib/libbinder.so (_ZN7android14IPCThreadState8transactEijRKNS_6ParcelEPS1_j+124)
  native: #05 pc 0001a8a3  /system/lib/libbinder.so (_ZN7android8BpBinder8transactEjRKNS_6ParcelEPS1_j+30)
  native: #06 pc 00088ce1  /system/lib/libandroid_runtime.so (???)
  native: #07 pc 00dbe539  /system/framework/arm/boot.oat (Java_android_os_BinderProxy_transactNative__ILandroid_os_Parcel_2Landroid_os_Parcel_2I+140)
  at android.os.BinderProxy.transactNative(Native method)
  at android.os.BinderProxy.transact(Binder.java:503)
  at m2.b$a$a.o0(:-1)
  at m2.e.S0(:-1)
  at x4.e.a(:-1)
  at r3.j$a.a(:-1)
  at e3.c.j(:-1)
  at e3.c.k(:-1)
  at e3.c.g(:-1)
  at e3.b.run(:-1)
  at android.os.Handler.handleCallback(Handler.java:743)
  at android.os.Handler.dispatchMessage(Handler.java:95)
  at android.os.Looper.loop(Looper.java:148)
  at android.app.ActivityThread.main(ActivityThread.java:5417)
  at java.lang.reflect.Method.invoke!(Native method)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:772)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:662)
```

evilmtv, 2022-03-22:
Hi @leslieam ,

As mentioned my app was closed before the crash happened for this crash log.

leslieam, 2022-03-22:
The crash happened at the time 03-14 14:46:16.399  which you mentioned.

The anr is another issue which not the same time slot.
```
----- pid 30228 at 2022-03-10 16:53:23 -----
Cmd line: sg.gov.tech.temi_changi_museum_app
```

zjn0505, 2022-04-01:
@evilmtv 

We have released the 128 version, there's a new way of adb connection in the Settings -> Developer Tools.

It is recommended to use it if you don't want to have the protected connection, meaning only you can connect to the robot wirelessly.

Or you are under a network without internet connection, then this the new and traditional way to connect.

evilmtv, 2022-04-04:
Hi @zjn0505 

Thank you for the update!
This is much needed for our environment.

evilmtv, 2022-04-04:
@leslieam 

My assumption is that the failed attempts to connect to Temi Center servers in a wifi without internet is consuming resources and causing the lag/crashes experienced as they do not happen in a wifi with internet.

I would like to suggest a new feature that may solve this issue:
Have a "offline mode" option in settings,
  this will disable all polling/checking of internet and updates sent to Temi Center servers etc
  will let user know that most features will not work in this mode such as Temi Center, NLP and voice..
  *different from Airplane mode which disables wifi etc as they should still be functional
On boot, instead of forcing the user to connect to a wifi that has internet, have an option to start in "offline mode" too.
  if Temi was already in offline mode before reboot, skip this wifi prompt entirely

leslieam, 2022-04-06:
@evilmtv 
Generally, our robot is designed to work with the internet. But some features can still work without the internet, Like navigation.
We fix some bugs when there's no internet. You can use the 128 version and see if the issue still happens.

evilmtv, 2022-06-05:
Hi!
Recently had a crash again and system became sluggish when switching back to a wifi without internet. It's on the latest firmware throughout.
Was working fine for a few weeks on a wifi with internet, smooth and without any crashes but the day we switched it back to the closed wifi without internet the system started becoming slow;

- Local video playback (in our own android app) took very long to start
- Temi Settings UI became slow
- Eventually Temi launcher/OS crashed and the "Starting..." page showed up which took quite awhile before it prompted for a wifi.

I understand that it is designed to work with internet as some features requires Temi servers to work, however for our use case we do not need those features and it specifically needs to be on a network that has internet access blocked to communicate with other devices.

Would be great if the offline mode I suggested earlier could be implemented.

leslieam, 2022-06-06:
Can you get us the logs?
`adb shell bugreport > bugreport.txt`

evilmtv, 2022-06-06:
Will get the logs when it happens again.
Also this is an example of the UI becoming slow:
https://youtu.be/B5fBT1M0v5k

leslieam, 2022-06-06:
I see it. 
But we can not know why without logs.

evilmtv, 2022-06-08:
As attached
[bugreport.txt](https://github.com/robotemi/sdk/files/8858028/bugreport.txt)

leslieam, 2022-06-08:
From the log, it seems your app is ANR and it is using most cpu.
 Can you please check what is your app is doing and is there anything taking a log of resource and time which result in the ANR.

[image]

evilmtv, 2022-06-08:
It should be using CPU since it's playing videos and a bunch of logic.

However these crashes do not occur when on a wifi with internet, our app (exactly the same) performs fine and Temi launcher/OS does not experience any slowdowns and crashes/ANRs.
The app can run continuously and smoothly for a few weeks without closing/restarts if on a wifi with internet.

Temi laucher/OS slowdown happens even when our app is not running (not even installed) when on a wifi without internet.

leslieam, 2022-06-08:
We'll look into it after we finish our high priority tasks.

evilmtv, 2022-06-08:
> We'll look into it after we finish our high priority tasks.

Awesome! I'll try to dig up more logs and data next week, hopefully it'll help to pinpoint the issue better.

evilmtv, 2022-07-12:
Hi @leslieam 
Captured some videos to further demonstrate the issue and that it is not due to our app.

1. https://youtu.be/lNZVS3vHx0M
This is what we experience often, sometimes leading to Temi launcher/OS crashing and restarting by itself.

2. https://youtube.com/shorts/7WAIZFIp3ik?feature=share
This is after uninstalling all of our apps, list of apps installed on Temi before this second clip shown below.
The laggy behaviour still exists despite removing our apps.
```
% adb shell pm list packages -f
package:/data/app/com.roboteam.teamy.usa-2/base.apk=com.roboteam.teamy.usa
package:/data/app/com.roboteam.teamy.videodemo.usa-1/base.apk=com.roboteam.teamy.videodemo.usa
package:/data/app/YouTube/YouTube.apk=com.google.android.youtube
package:/system/priv-app/TelephonyProvider/TelephonyProvider.apk=com.android.providers.telephony
package:/system/priv-app/CalendarProvider/CalendarProvider.apk=com.android.providers.calendar
package:/system/priv-app/MediaProvider/MediaProvider.apk=com.android.providers.media
package:/system/priv-app/WallpaperCropper/WallpaperCropper.apk=com.android.wallpapercropper
package:/system/app/DocumentsUI/DocumentsUI.apk=com.android.documentsui
package:/system/app/Galaxy4/Galaxy4.apk=com.android.galaxy4
package:/system/priv-app/ExternalStorageProvider/ExternalStorageProvider.apk=com.android.externalstorage
package:/system/app/HTMLViewer/HTMLViewer.apk=com.android.htmlviewer
package:/data/app/com.roboteam.teamy.iheart.usa-1/base.apk=com.roboteam.teamy.iheart.usa
package:/system/priv-app/MmsService/MmsService.apk=com.android.mms.service
package:/system/priv-app/DownloadProvider/DownloadProvider.apk=com.android.providers.downloads
package:/system/app/WinStartService/WinStartService.apk=com.android.winstart
package:/system/app/SoundRecorder/SoundRecorder.apk=com.android.soundrecorder
package:/system/priv-app/DefaultContainerService/DefaultContainerService.apk=com.android.defcontainer
package:/data/app/com.android.vending-1/base.apk=com.android.vending
package:/system/app/PacProcessor/PacProcessor.apk=com.android.pacprocessor
package:/system/app/CertInstaller/CertInstaller.apk=com.android.certinstaller
package:/system/framework/framework-res.apk=android
package:/system/app/Camera2/Camera2.apk=com.android.camera2
package:/system/priv-app/BackupRestoreConfirmation/BackupRestoreConfirmation.apk=com.android.backupconfirm
package:/system/app/Provision/Provision.apk=com.android.provision
package:/system/priv-app/StatementService/StatementService.apk=com.android.statementservice
package:/system/app/HoloSpiralWallpaper/HoloSpiralWallpaper.apk=com.android.wallpaper.holospiral
package:/system/app/PhaseBeam/PhaseBeam.apk=com.android.phasebeam
package:/data/app/com.roboteam.teamy.knowledgeskill.usa-1/base.apk=com.roboteam.teamy.knowledgeskill.usa
package:/system/priv-app/SettingsProvider/SettingsProvider.apk=com.android.providers.settings
package:/data/app/com.roboteam.teamy.camera.usa-1/base.apk=com.roboteam.teamy.camera.usa
package:/system/priv-app/SharedStorageBackup/SharedStorageBackup.apk=com.android.sharedstoragebackup
package:/system/app/PrintSpooler/PrintSpooler.apk=com.android.printspooler
package:/system/app/BasicDreams/BasicDreams.apk=com.android.dreams.basic
package:/system/priv-app/InputDevices/InputDevices.apk=com.android.inputdevices
package:/data/app/com.robotemi.browser-1/base.apk=com.robotemi.browser
package:/system/app/TemiInstall/TemiInstall.apk=com.luckystar.luckstarautoinstall
package:/data/app/com.google.android.webview-2/base.apk=com.google.android.webview
package:/system/priv-app/OneTimeInitializer/OneTimeInitializer.apk=com.android.onetimeinitializer
package:/system/priv-app/Telecom/Telecom.apk=com.android.server.telecom
package:/system/app/KeyChain/KeyChain.apk=com.android.keychain
package:/data/app/com.google.android.gms-2/base.apk=com.google.android.gms
package:/system/priv-app/GoogleServicesFramework/GoogleServicesFramework.apk=com.google.android.gsf
package:/data/app/GoogleTTS/GoogleTTS.apk=com.google.android.tts
package:/system/priv-app/PackageInstaller/PackageInstaller.apk=com.android.packageinstaller
package:/system/app/PicoTts/PicoTts.apk=com.svox.pico
package:/system/app/KedmaneeInput/KedmaneeInput.apk=com.keyboard.android.apps.inputmethod.thai.kedmanee
package:/data/app/io.firedome.temi-1/base.apk=io.firedome.temi
package:/system/priv-app/ProxyHandler/ProxyHandler.apk=com.android.proxyhandler
package:/system/app/LatinIME/LatinIME.apk=com.android.inputmethod.latin
package:/system/priv-app/ManagedProvisioning/ManagedProvisioning.apk=com.android.managedprovisioning
package:/system/app/MediaFloat/MediaFloat.apk=com.android.rk.mediafloat
package:/system/app/PhotoTable/PhotoTable.apk=com.android.dreams.phototable
package:/system/app/NoiseField/NoiseField.apk=com.android.noisefield
package:/system/priv-app/GoogleLoginService/GoogleLoginService.apk=com.google.android.gsf.login
package:/system/app/WAPPushManager/WAPPushManager.apk=com.android.smspush
package:/system/app/LiveWallpapersPicker/LiveWallpapersPicker.apk=com.android.wallpaper.livepicker
package:/system/priv-app/StressTest/StressTest.apk=com.cghs.stresstest
package:/system/app/KoreanInput/KoreanInput.apk=com.google.android.inputmethod.korean
package:/system/priv-app/Settings/Settings.apk=com.android.settings
[... 3 lines cut ...]
package:/data/app/com.roboteam.teamy.newsskill.usa-1/base.apk=com.roboteam.teamy.newsskill.usa
[... cut ...]
package:/system/priv-app/ContactsProvider/ContactsProvider.apk=com.android.providers.contacts
package:/system/app/CaptivePortalLogin/CaptivePortalLogin.apk=com.android.captiveportallogin
package:/data/app/com.roboteam.teamy.youtubeskill.usa-1/base.apk=com.roboteam.teamy.youtubeskill.usa
```

3. https://youtube.com/shorts/htNnhxMKzrk?feature=share
Another clip showing slow issues.

4. https://youtube.com/shorts/KrzN8fHgQuI?feature=share
If wifi is turned off this issue does not occur anymore. (Also true if connected to a wifi with internet access)

evilmtv, 2022-07-12:
Tested with two different Temi too and same issue on both.

leslieam, 2022-07-18:
@zjn0505 
please take a look on this.

evilmtv, 2022-08-30:
Hi @leslieam @zjn0505,
Are there any updates on this?

We've tested on Temi v3 with the latest SDK yesterday and the issues affecting us when on a wifi without internet still persists (freeze and crashes).
Everything is fine when on a wifi with internet; general temi usage as well as using our custom tour app.

closed 2024-03-26 by evilmtv