robotemi/sdk #267: New ADB connection not working

Bjastkuliar, 2022-02-10:
I've been working on my temi for a while,
since the "last" update to Temi ADB's feature I can no longer connect my laptop to it.

Before, when using the CLI and typing "adb connect <temi's IP>:5555 I was able to connect to it without any problem at all.

Now that you are required to use its generated password (going to the URL via browser or via curl) I can no longer access it, it's not an issue of the Temi itself because my colleague is able to connect to it. When I try to reach the link the only output I get is:

{
  "status": "Failed",
  "error": "Error Code 02"
}

I'd like to know what this "Error Code 02" is, unfortunately the documentation looks to be quite unclear about it, is there a place where to check the error codes? I've tried disabling the Firwall, adding Android Studio/ adb to its exceptions, without success...

leslieam, 2022-02-11:
What is the serial number of your robot?
We can send you a temporary fix for that.

Bjastkuliar, 2022-02-11:
> What is the serial number of your robot? We can send you a temporary fix for that.

Hi, so the serial number is 00121185544
But I'd much rather know what the issue itself is and what the error code means, because (as I mentioned before) my colleague is able to connect from its Mac, while I (from my windows laptop) cannot. There must be some connection issues between my laptop and temi...

Kind Regards

closed 2022-02-11 by leslieam