robotemi/sdk #270: List of description IDs and descriptions for onGoToLocationStatusChangedListener

evilmtv, 2022-03-08:
Hi,

Is there a list available somewhere for the list of possible IDs and descriptions?
Could not find more info in the wiki docs.
https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener

lcgao, 2022-03-09:
|ID|Description|
|-|-|
|1060|"Path Plan"|
|2000|"Obstacle"|
|2002|"Hight Obstacle"|
|2003|"Lidar Obstacle"|
|2004|"Front Obstacle"|
|2005|"Back Obstacle"|
|2006|"Abyss Obstacle"|
|2007|"Virtual Wall Obstacle"|
|2008|"Cliff Detected"|
|2009|"Stuck Wheel"|
|-|"Calculating"|

chesterkwan, 2022-03-13:
Hi, 

I have received some abort events with description id of 1005 and 0. And both of them have the description 'unknown'.
At first, I thought that 1005 means a tap on the head button to abort the movement. But we have a Temi with the head button disabled but it still reports 1005 abort. I am really confused with the description id.

I would also like to know, if the id number is following some sort of pattern - like do the first digit 2 means obstacle and 1 means abort? 

Thanks in advance.

zjn0505, 2022-03-29:
@chesterkwan 

Please check the updated reply above.

1005 means there is a stop command triggered by user, or by internal/external logic, including the `stopMovement` SDK interface.

We will update the doc and supporting software for clarification.