robotemi/sdk #273: Failing to connect through ADB

Ryan-Lee-2000, 2022-03-15:
I'm trying to connect to my Temi using my mobile hotspot but it keeps giving me this error:
curl: (7) Failed to connect to 192.168.234.143 port 5556 after 3167 ms: Connection refused

When I tried using the browser instead I get "Site not found" connection refused.

chesterkwan, 2022-03-15:
Same here, still working last night.

zjn0505, 2022-03-15:
Please try to reboot the robot and retry adb connection.
We had a [service maintenance](https://status.robotemi.com) yesterday, it would affect the authentication of trust devices. Please try again and let us know the result.

SlevinKlvra, 2022-03-15:
Hello. We have the same issue here. We are not able to connect with temi by adb connection. We already have tried to reboot the robot but still doesn't connecting. Thank you

paxbo, 2022-03-15:
Same Issue here. Even after reboot it doesn't work.

Ryan-Lee-2000, 2022-03-15:
@zjn0505 Hi, I've rebooted the Temi but I'm still encountering the same issue. I've tried to connect to another seperate Temi and faced the same issue as well.

zjn0505, 2022-03-15:
We are working on it, will update this thread to let you know.

eSalto, 2022-03-15:
We still have this issue and can't connect to Temi by adb.
Therefore, we can't debug our code.
I understand that the bug is still active?
Thanks.

sduenasg, 2022-03-15:
Same issue here... we also did a hard reset, the issue persists.

zjn0505, 2022-03-16:
We deployed a fix on the cloud.
If you still have issue, please send us your serial number.
We are working on it to make it more stable and suitable for offline adb connection.

eSalto, 2022-03-16:
> We deployed a fix on the cloud.

The issue has been resolved for us since early in the morning.
Thanks.

MarcTestier, 2022-03-17:
> We deployed a fix on the cloud. If you still have issue, please send us your serial number. We are working on it to make it more stable and suitable for offline adb connection.

Hi, we had this issue too and it got fixed for the Temis that were already setup before.
We just tried to setup to 2 new Temis today, just fresh out of the box and we're having this "Connection refused" error on these 2 Temis only.

How may we contact you to provide the serial number?

Thanks.

zjn0505, 2022-03-17:
> Hi, we had this issue too and it got fixed for the Temis that were already setup before. We just tried to setup to 2 new Temis today, just fresh out of the box and we're having this "Connection [...]

The workaround we deployed earlier doesn't apply for new temis, they can't get recognized as trusted devices before we eventually fix this issue.
You can email me to charlie.zhang [at] robotemi.com. We will address it asap.

zjn0505, 2022-04-01:
We have released the 128 version, there's a new way of adb connection in the Settings -> Developer Tools.

It is recommended to use it if you don't want to have the protected connection, meaning only you can connect to the robot wirelessly.

Or you are under a network without internet connection, then this the new and traditional way to connect.

closed 2022-04-01 by zjn0505