robotemi/sdk #275: getMapList and getAllSequences returning empty lists

sduenasg, 2022-03-22:
Hi,

I'm trying to get the map list and the sequences list. I have two sequences and two maps on Temi Center. I added the following permissions to the manifest:

```
<meta-data
            android:name="@string/metadata_permissions"
            android:value="com.robotemi.permission.settings, com.robotemi.permission.sequence, com.robotemi.permission.map" />

```

I also request the permissions in the onRobotReady function like this:

```
if (isReady) {
    Robot.getInstance().addOnRequestPermissionResultListener(this)
    Robot.getInstance().requestPermissions(listOf(Permission.SEQUENCE, Permission.MAP), 2)
...
}
```

The listener reports the permissions are Granted, and I see it in Temi's permissions settings menu that the app has these permissions.

getMapList() and getAllSequences() still return empty lists though.

Any ideas?

lcgao, 2022-03-24:
Hi, please also make sure the SDK is ready before invoking these two functions. Listen to the listener OnRobotReadyListener to judge if the SDK is ready.

sduenasg, 2022-03-24:
Thank you. Yes, we are invoking after the SDK is ready, both list turn out empty.

```
override fun onRobotReady(isReady: Boolean) {
  if (isReady) {
  (...)
      Robot.getInstance().addOnRequestPermissionResultListener(this)
      Robot.getInstance().requestPermissions(listOf(Permission.SEQUENCE, Permission.MAP), 2)
  (...)
  
      val seqList = Robot.getInstance().getAllSequences()
      val mapList = Robot.getInstance().getMapList()
  }
}
```

lcgao, 2022-03-24:
It is weird, could you please try to test with the SDK Sample? In SDK Sample, you can find the corresponding buttons to get the seqeunces and map list.

sduenasg, 2022-03-24:
Hi, we just tried the sample app in the sdk . The "get sequences" button is not working:

[image]

The "get maps" one doesn't seem to do anything either.

zjn0505, 2022-03-26:
We are checking get all sequences API. We will let you know once it is fixed.

Meanwhile I checked getMapList API, and haven't seen error reported on our backend. There is a throttle control at the backend against this API, but it should affect if you just try once or twice.

Can you leave your SN here or send to charlie.zhang [ at ] robotemi.com? We will take a closer look with this info.

sduenasg, 2022-03-28:
Thank you. The serial number is: 00121185550

The getMapList keeps returning empty lists in our app and in the sample one. We have 3 maps saved on Temi Center:

[image]

zjn0505, 2022-03-30:
We've deployed the fixes. I think you can get the sequence saved online. Currently I see 1 sequence shall be returned.

However, I still cannot see any issue from my side regarding getMapList API. From the API call I see there should be 3 maps returned.

Please feel free to contact me via email. We can arrange a debug session with you.

closed 2022-03-30 by sduenasg