robotemi/sdk #277: setLocked(boolean) and isLocked() not working

sduenasg, 2022-03-30:
Hi again, we are trying to toggle locked mode (password lock) via the SDK using these:

https://github.com/robotemi/sdk/wiki/Utils#setLocked

But it isn't working, they show up as an unresolved reference in Robot.getInstance(). Is there any workaround for this?

lcgao, 2022-03-31:
Hi, this function is supported from version 0.10.74 as the wiki documentation mentioned. So please update the SDK version in app/build.gradle and sync to use this function.

sduenasg, 2022-03-31:
We are using the latest sdk 0.10.79 at the moment. I've seen in the sample app that there is a direct way to read/write Robot.getInstance().locked variable. I'll try tomorrow (I'm not with Temi today) and close this issue if it works. Thanks!

sduenasg, 2022-04-01:
it is working, thanks!

closed 2022-04-01 by sduenasg