robotemi/sdk #278: Overview of user peer-IDs

LeoFau, 2022-04-13:
Hello,

is there anywhere I can access the peer-ID of every user without displaying all the user information programmatically?
Maybe in the temi center or in the app?
I couldn't find it so far and would like to set a few fixed contacts for telepresence.

Thank you in advance.

lcgao, 2022-04-14:
Hello, do you want to get peer-IDs of all user programmatically? If yes, using [`Robot.getInstance().getAllContact()`](https://github.com/robotemi/sdk/wiki/Users-&-Telepresence#getAllContact).

LeoFau, 2022-04-14:
Hello,
as stated in the original question, I would like to find this information without integrating this into my code. 
I thought that this information is probably available elsewhere. For example in the temi center or the temi app. As admin I thought I would be able to access this information somewhere without running this code (which I know works, as I have used it before, but I am looking for an alternative).

Thank you

lcgao, 2022-04-15:
Hello, please open a ticket for this new request on Zendesk. Thanks.

zjn0505, 2022-04-15:
@LeoFau 
As this is not a user oriented feature, and the contact peer id is quite meaningless to be displayed on UI, we cannot show it on temi center or mobile for regular user access.

However, it is accessible for developers to find it out on temi center.

I would suggest to use temi center with browser dev tool opened, open the members page of a robot, and search in the network tab with filtering of "members".

`clientId` there is what we are looking for.

[image]

LeoFau, 2022-04-20:
Perfect, thank you!

closed 2022-04-20 by LeoFau