robotemi/sdk #279: Temi STT problem

SorcererTimz, 2022-04-21:
Can Temi use ASR input word to app?

lcgao, 2022-04-27:
Hi, you mean ASR input method or just want to get the ASR result after waking up temi in your App?

zjn0505, 2022-04-28:
ASR as voice input, right?

zjn0505, 2022-04-28:
It can be achieved but not easily, as ASR is part of voice interaction in current design.
You will need to manipulate the voice flow inside your app to use ASR as voice input, including

- hide conversation layer.
- trigger wake up from SDK.
- override NLP to intercept ASR result.

I think we don't have dynamic control over conversation layer and NLP overriding yet if user finish voice input and continue with regular voice interaction.

SorcererTimz, 2022-05-17:
> Hi, you mean ASR input method or just want to get the ASR result after waking up temi in your App?

Yes i want to use Temi ASR as a input method

SorcererTimz, 2022-05-17:
OR is there Temi able to call the google Text to speech API as a input method?

yausum, 2022-05-23:
I think he would like to know, can he enable a speak to text function by using the Temi SDK? 
Since he want to tailor made an application on Temi, with a speak to text feature. So that when user trying to use his application on Temi, they don't have to type the words anymore.