robotemi/sdk #281: onAsrResult is not working after overriding STT and conversation layer

iamkhan001, 2022-04-27:
Hi there,

I'm working on chat bot application in temi, where i need to recognise speech without using wakeup word. 

following is manifest settings i have added

```
<meta-data android:name="com.robotemi.sdk.metadata.KIOSK" android:value="true" />
<meta-data android:name="@string/metadata_override_conversation_layer" android:value="true" />
<meta-data android:name="@string/metadata_override_stt" android:value="true" />
<meta-data android:name="@string/metadata_override_nlu" android:value="true" />
```

in activity once robot initiate i added following code

```
    override fun onRobotReady(isReady: Boolean) {
        Log.e(TAG, "onRobotReady $isReady")
        if (isReady) {
            try {
                refreshTemiUi()
                robot.wakeup()
                Log.e(TAG,">>> Wake up <<<")
            } catch (e: PackageManager.NameNotFoundException) {
                throw RuntimeException(e)
            }
        }
    }
```
when user said wakeup word, i can see logs inside `onWakeupWord` but i cannot see any logs for `onConversationStatusChanged` also no logs for `onAsrResult` to get what customer is saying. 

please help on this.

lcgao, 2022-04-27:
Hi, I saw you added the meta-data for override STT(Speech to Text aka ASR), it means temi will not trigger the default original ASR anymore and expecting the custom one that provided by the App. So there is no other log except inside onWakeupWord().

So if you still want to use the default ASR, remove the meta-data(override_stt), same goes for NLU.

iamkhan001, 2022-10-12:
Thanks @lcgao

closed 2022-10-12 by iamkhan001