robotemi/sdk #283: Temi can't find saved locations via goTo-method

LeoFau, 2022-04-28:
Hello,

I have a small issue with my temi's goTo-method. Everytime I try to send him somewhere with the method, he seems to know the location (Screen says "Going to <locationname>"), but then after some time I get an error that says he couldn't find the way to the location and that it might be outside of the map.
The robot is inside the parameters of the map and I made sure the String is the exact same as the locationname on the robot.

Is there something else I need to consider when using the method?

`robot.goTo("locationname");`

LeoFau, 2022-04-29:
When I send them to the same location via the locations-function on the robot itself he can drive there though.

evilmtv, 2022-05-06:
1. When you call the `goTo()` function from your app, Temi's screen shows "going to _" without the location name?
2. To verify the string is correct, pull the list of locations using this https://github.com/robotemi/sdk/wiki/Locations#getLocations

LeoFau, 2022-05-06:
1. No, it shows "going to <locationname>". The name shown on the screen is the correct location name.
2. I did double check the locations list and the locations' names are exactly the same as the input to the method I tried. Only that they are all in lower case letters, but in your sample code it shows, that the goTo-method converts the entered String to lower case anyways, so that shouldn't be the problem.

lcgao, 2022-05-07:
Hi, please report this issue on Zendesk, thanks.

lcgao, 2022-05-07:
Hi, what is the log (from [OnGoToLocationStatusChangedListener](https://github.com/robotemi/sdk/wiki/Locations#onGoToLocationStatusChangedListener)) showing after calling goTo-method?