robotemi/sdk #287: why bindService is always fl

jackk2347, 2022-06-06:
[image]

lcgao, 2022-06-06:
It means SDK didn't conntect with temi Launcher successfully. The App using temi SDK should run on real temi robot instead of Android simulator.

jackk2347, 2022-06-06:
Oh! you mean that if the apps is running in the temi is working? it cannot running in virtual device?

lcgao, 2022-06-06:
Yes, actually temi SDK dependents on temi Launcher App, so temi Apps cannot run on a virtual device that doesnot install temi Launcher.

jackk2347, 2022-06-06:
Also, i want to ask that the used api is  just need implementation 'com.robotemi:sdk:0.10.80' right?

lcgao, 2022-06-06:
Yes, we just provid this library.

closed 2022-06-10 by lcgao