robotemi/sdk #289: how to use the asrlistner function , can give me some example ?

jackk2347, 2022-06-08:
how to use the asrlistner function , can give me some example ?
for example : how to addAsrListener and use the  onAsrResult function? 
[image]

lcgao, 2022-06-08:
Please refer to the Sample code [here](https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt).

jackk2347, 2022-06-08:
however , where is received the voice message?

lcgao, 2022-06-08:
The callback method `onAsrResult()` you implements from the `AsrListener`.

jackk2347, 2022-06-08:
Ok, the onAsrResult() is received the voice meassge  first and response it? However , i don know how to received the voice message first .

lcgao, 2022-06-08:
If you register the AsrListener correctly, the method onAsrResult() will be invoked by SDK and your App will be able to receive the voice message from there.

jackk2347, 2022-06-08:
Ok , thank a lot . it is working

jackk2347, 2022-06-08:
Also, i want to ask that can i set the voice message first and then wake up the temi?

closed 2022-06-10 by jackk2347