robotemi/sdk #292: can i say "Hi temi" first and then run the wakeup function . Like the iphone siri .

jackk2347, 2022-06-09:
can i say "Hi temi" first and then run the wakeup function . Like the iphone siri .

lcgao, 2022-06-10:
Say "Hi temi" has the same effect as call function `wakeup()`, so no need to wake up temi twice via two different way.

jackk2347, 2022-06-10:
However , i say "Hi temi" after then it don't depends on my onAsrResult to response me . How to solve it?

lcgao, 2022-06-10:
The flow is:
1. User says "Hi temi".
2. temi is waked up by "Hi temi" and waiting for the user's voice input.
3. If user says "How are you", the function onAsrResult() will be called by SDK and the input argument will be "How are you".
4. It means your App is able to receive the voice input by onAsrResult() and do some response.

jackk2347, 2022-06-10:
it just depends on defalt onAsrResult to response me , it is not override the method .

jackk2347, 2022-06-10:
Also, can i change the wakeup word . Like ""Hi temi" to change "Hi robot " to wakeup temi?

lcgao, 2022-06-10:
I guess you don't want temi respond you by default reponse but by your custom response? If so, plese refer to the tutorial [here](https://github.com/robotemi/sdk/wiki/Speech#override-the-nlp-).

lcgao, 2022-06-10:
Yes.

lcgao, 2022-06-10:
By the way, please consider using the custom Q&A. User can config the Q&A on temi Center and let temi respond according to the configuration.

jackk2347, 2022-06-10:
Also, i want to ask the one question . how to disable the wakeup function on the start . For example : i speak "Hei Temi" then temi cannot response me .  Also , i want to make the button to able the temi wakeup function . Can i make it?

lcgao, 2022-06-10:
https://github.com/robotemi/sdk/wiki/Utils#toggleWakeup