robotemi/sdk #293: Can i disable the wakeup function on the apps start?

jackk2347, 2022-06-10:
Also, i want to ask the one question . how to disable the wakeup function on the start . For example : i speak "Hei Temi" then temi cannot response me . Also , i want to make the button to able the temi wakeup function . Can i make it?

evilmtv, 2022-06-14:
You can disable wakeup function manually in settings, or programmatically via
https://github.com/robotemi/sdk/wiki/Utils#toggleWakeup

On your app you can add a button to trigger wakeup via
https://github.com/robotemi/sdk/wiki/Speech#wakeup

jackk2347, 2022-06-14:
i want to ask one more question . when i open the klosk mode first then it can disable the wakeup function . However , when i turn on the wakeup funation then it cannout base on my asrresult function to response me . i think that it may about the klosk mode is overrider my asrresult and i just can see the onwakeupword has the result on the log  message . Also , it do not have the temi speak message to show out to record my vocie .