robotemi/sdk #299: Kiosk mode and built-in orders issues

sduenasg, 2022-06-21:
Hello. In order to activate Greet mode from our app, we are activating Kiosk Mode too (sdk restriction)

Thing is, when Kiosk mode is enabled (and our app is the home screen) the robot doesn't respond to any built-in orders (like "go to") nor any other Q&A queries. After saying "hey temi" and the order, it just sits there moving its eyes up and down. Disabling kiosk mode solves this issue. But we have to activate it to be able to activate greet mode from the app, which is one of our requirements.

Any ideas?

SDK version is 0.10.80

zjn0505, 2022-06-22:
In 128 version with temi launcher app 16398, turn on Greet Mode from SDK cannot bypassed and requires kiosk mode.

sduenasg, 2022-06-22:
Thank you. For some reason (probably testing stuff), our app was overriding NLP. It's fixed now.

closed 2022-06-22 by sduenasg