robotemi/sdk #302: Confusion On ASRListener and Temi Speach

shahzib111, 2022-06-28:
I am working with the temi speech api in kiosk mode and while working with the temi sample app, and although the onAsrResult does get called on voice, I had a couple questions in regards to this. First, Lets say I start a conversation with a user using the following:
`            robot.askQuestion("welcome, To get started, may i have your full name")`

and in response to this the user says "hi, my name is david", the line from the onAsrResult function:
`asrResult.contains("hi my name is", ignoreCase = true) -> {
                robot.askQuestion("Hello, I'm temi, what can I do for you?").        //this line is not called.
            }`
should be called but is never called in my case. How can I process the response from the question above and ask the next question  and process that response. In other words, how can we process user commands based on the questions the program/temi asks the user and move to the next one for the same thing.

my onAsrResult function is:
```
override fun onAsrResult(asrResult: String) {
        printLog("onAsrResult", "asrResult = $asrResult")
        try {
            printLog("here", "here = here")

            val metadata = packageManager
                    .getApplicationInfo(packageName, PackageManager.GET_META_DATA).metaData
                    ?: return
            if (!robot.isSelectedKioskApp()) return
            if (!metadata.getBoolean(SdkConstants.METADATA_OVERRIDE_NLU)) return
        } catch (e: PackageManager.NameNotFoundException) {
            printLog("error", "error = error")

            e.printStackTrace()
            return
        }
        when {
            asrResult.contains("hi my name is", ignoreCase = true) -> {
                robot.askQuestion("Hello, I'm temi, what can I do for you?").        //this line is not called.
            }
            asrResult.equals("Play music", ignoreCase = true) -> {
                robot.finishConversation()
                robot.speak(create("Okay, please enjoy.", false))
            }
            asrResult.contains("test", ignoreCase = true) -> {
                robot.askQuestion("Hello, I'm temi, what can I do for you?")
            }
            asrResult.equals("Play movie", ignoreCase = true) -> {
                robot.finishConversation()
                robot.speak(create("Okay, please enjoy.", false))
            }
            asrResult.toLowerCase(Locale.getDefault()).contains("follow me") -> {
                robot.finishConversation()
                robot.beWithMe()
            }
            asrResult.toLowerCase(Locale.getDefault()).contains("go to home base") -> {
                robot.finishConversation()
                robot.goTo("home base")
            }
            else -> {
                robot.askQuestion("Sorry I can't understand you, could you please ask something else?")
            }
        }
    }
```

lcgao, 2022-06-28:
Hi, `"hi, my name is david".contains("hi my name is", ignoreCase = true")` would be `false`, note the comma ',' after 'hi'.

shahzib111, 2022-06-29:
It doesnt seem to matter. Temi catches the voice as hi my name is david as is printed out on the screen but instead of responding with "Hello, I'm temi, what can I do for you?"  she goes to the default onAsrListener and responds "Hi, my name is Temi"

ShehryarZaheer, 2025-02-01:
Hi, were you able to find a solution for this?