robotemi/sdk #309: Is it possible that Temi can navigate between floors?

eSalto, 2022-07-19:
I am trying to get Temi to navigate on different floors. I have seen in the Temi SDK sample that there is a loadFloorAtElevator function. I was wondering if it is implemented or still under development.

zjn0505, 2022-07-20:
Thanks for asking.

There is no out-of-box capability of temi to call and ride the elevator, as it is up-to the local regulations and elevator specs.

But it is possilbe for current temi to store multiple maps/floors in the disk and load them dynamically on need.

You can build multi-floor maps with temi, and load the maps with temi SDK.

----

https://github.com/robotemi/sdk/wiki/Locations#loadMapWithPosition

The method above provides interface to load map by map id, with optional parameters auto-reposition and position of target map.

https://github.com/robotemi/sdk/blob/master/sdk/src/main/java/com/robotemi/sdk/Robot.kt#L2656-L2700

Here in the code lists all the methods that will help you navigate between floors.

eSalto, 2022-07-20:
Perfect, I'll take a look at it. Thank you very much

eSalto, 2022-07-20:
A question appears when loading a map on the screen that is loading the map. Can it be removed or is it necessary?

zjn0505, 2022-07-21:
In 128 version and before, there will be a blocking screen pops up while loading a map.

This is a protection from unwanted map editing and navigation when robot is replacing its map.

We will provide an argument in the loadMap interface to not showing the pop-up. Then it is the responsibility of the app to protect the map switching phase.

eSalto, 2022-07-21:
Perfect, I'll wait for the new version.

eSalto, 2022-07-21:
Is there any way to relate MapModel to Floor? I wanted to do it with mapId but I have seen that the id of the MapModel does not match.

zjn0505, 2022-07-22:
They are different ids. There is such complexity that we have to use separated identifiers on map for different purpose.

Id in [MapModel](https://github.com/robotemi/sdk/wiki/Locations#mapmodel) is generated from temi cloud when you create a map back up.

You can get a list of maps you have backed-up via [getMapList()](https://github.com/robotemi/sdk/wiki/Locations#getMapList), and load a map with that id via [loadMap()](https://github.com/robotemi/sdk/wiki/Locations#loadmap)

You can back-up the same map many times, and their ids in MapModel will be different.

On the other hand, id in Floor is not documented and it shall almost only be used for multi-floor purpose which we haven't made it ready for public yet. It is generated from the mapping algorithm on the robot.

Floor has similiar data as [MapDataModel](https://github.com/robotemi/sdk/wiki/Locations#mapdatamodel), which is an abstraction of the map data running on temi. The map id in Floor and MapDataModel is unqiue for each physical maps calculated by algorthm.

You can back-up the same map many times, and add different locations, walls, pathes to them, but their ids in Floor and MapDataModel will be the same.

zjn0505, 2022-07-23:
You can already use current SDK and robot to achieve quick switch between different maps. There will be up to 32 maps cached locally if you bulit them and have them backup from robot to temi center, or import from cloud to robot.

From the product decision that we don't want to make UI and some storage logic of multifloor publicly available yet. 

If you have a specifc scenario that really depends on multifloor, you can reach out to your distributor for this feature and they can help you setup.