robotemi/sdk #311: robot.locations returns empty string

aparajita13, 2022-07-20:
Hi,
I am currently working on Temi robot to send it to the locations already saved on the robot. However, when I try to call robot.goTo(location) it stays still. On checking the list of locations by printing robot.locations, I see no saved locations. I have two saved location on Temi. Is there something that needs to be imported (apart from temi sdk) for see the list of locations on the robot?

Thanks.

[image]

[image]

zjn0505, 2022-07-21:
I suppose it is due to the timing you initiate SDK interfaces.

I would suggest not to initiate locations together with activity, cause the conenction and communication is built after your app launched in [TemiSdkContentProvider](https://github.com/robotemi/sdk/blob/master/sdk/src/main/java/com/robotemi/sdk/TemiSdkContentProvider.kt)

You would need to wait for the log "onServiceConnected" to start to asking for locations. Without the connection and service running, it will return empty locations list

https://github.com/robotemi/sdk/blob/master/sdk/src/main/java/com/robotemi/sdk/Robot.kt#L1039

So you can change `val location = Robot.getInstance().locations` into a lateinit or lazy variable. You also should not use it as a `val`, because the locations would be altered when your app is running, e.g. from temi UI, temi center, or from your app. Using a `val` would create an immutable instance, and you won't get the up-to-date location list.

aparajita13, 2022-07-21:
Thanks so much. I have created a separate java class and introduced some delay, now it works.

closed 2022-07-22 by zjn0505