robotemi/sdk #313: Get temi location from temi Center

cyrsis, 2022-07-27:
Is there way to have a history of temi location from api instead of robot itself?

zjn0505, 2022-07-27:
I am not fully clear about what data is expected.

Can you specify which SDK interface shall be the counterpart you want?

[getLocations()](https://github.com/robotemi/sdk/wiki/Locations#getLocations) will provide the names of current map.

[getMapData()](https://github.com/robotemi/sdk/wiki/Locations#getMapData) will give you the entire map, lit has ocations with coordinates included.

We don't provide cloud APIs yet. I guess you want to get the locations from you cloud? You may be able to check Chrome console and get the network request for the relavent data, but it only has a short life-span token and not suitable for long time use.

cyrsis, 2022-09-21:
hopefully one day u guys would have some service token available for the mass temi deployment

closed 2022-09-21 by cyrsis