robotemi/sdk #314: 想問有沒有方法可以在speak 打後 , 確定他的話語完畢,在開始下一句?

jackk2347, 2022-07-28:
想問有沒有方法可以在speak 打後 , 確定他的話語完畢,在開始下一句? 我現在的做法是把幾句話語放一個stack 上 , 在確定他的話語狀態是"COMPLETED"的時侯pop 出黎 , 在想完畢後在開始下一句 , 但是他只執行一次 onTtsStatusChanged 這個method , 有沒有方法可以在每一次話語後 , 都回到 onTtsStatusChanged 這個method

zjn0505, 2022-07-28:
Hi your design should work. There are already many use cases like this, and in temi we are doing similiar things to chain TTS in the sequence feature.

Every Robot.speak() will trigger a chain of callbacks in onTtsStatusChanged as PROCESSING -> STARTED -> COMPLETED.

id in TtsRequest from these callbacks shall be the same as the one you send in Robot.speak().

jackk2347, 2022-07-28:
有什麼例子嗎?

zjn0505, 2022-07-29:
目前沒有即時可用的例子。

Every TtsRequest should end with a status of COMPLETED or ERROR in onTtsStatusChanged callback.

That is the moment you can pop/poll from the stack/queue for next speech and pass it to Robot.speak() for next iteration.

jackk2347, 2022-07-29:
> 目前沒有即時可用的例子。
> 
> Every TtsRequest should end with a status of COMPLETED or ERROR in onTtsStatusChanged callback.
> 
> That is the moment you can pop/poll from the stack/queue for next speech and pass it to Robot.speak() for next iteration.

我是這樣做的 , 但是當我在用 Robot.speak()的時候,他沒有回到 onTtsStatusChanged  這個method .

zjn0505, 2022-07-29:
Does the robot actually speak after Robot.speak()?

Can you try with temi SDK sample?

You can complie it yourself from the repository or download our pre-compiled version [here](https://app.bitrise.io/app/632e7a14a162151f/build/c2a9be9d-bf2c-48b7-b572-518d69db672b/artifact/44f21e36e10c1857/p/1d30d810c847a4c69407bee1ad86c05d)