robotemi/sdk #317: Can I know if telepresence is DECLINED?

luozhexin, 2022-08-08:
I checked the [OnTelepresenceStatusChangedListener](https://github.com/robotemi/sdk/wiki/Users-&-Telepresence_chn#onTelepresenceStatusChangedListener) .It is a abstarct class and SDK-sample no contains .So is this method feasible? If so, how can I change it to an interface or some other way to monitor.

zjn0505, 2022-08-10:
Telepresence callbacks are not well maintained, only `OnTelepresenceEventChangedListener` is working, but it won't give you enough info about the delined event. Only STARTED and ENDED can be received.

zjn0505, 2022-08-10:
The original design was, startTelepresence() should return a `sessionId` for you to initiate a listener as followed:

```
val sessionId = Robot.getInstance().startTelepresence(name, userId)

Robot.getInstance().addOnTelepresenceStatusChangedListener(object : OnTelepresenceStatusChangedListener(sessionId) {
    override fun onTelepresenceStatusChanged(callState: CallState) {
        when (callState.state) {
            CallState.State.STARTED -> {}
            CallState.State.ENDED -> {}
            CallState.State.DECLINED -> {}
            else -> {}
        }
    }
})
```

But after a redesign of the telepresence flow 2 years ago. sessionId is no longer returned, thus the callback cannot be initialized.

closed 2022-08-22 by zjn0505