robotemi/sdk #320: Can Temi slow down the moving speed during goTo command?

kakeung, 2022-08-15:
I developed an application to remote control Temi by SDK.

Test 1:
1. send `goTo()` command to Temi start from location A to B using `SpeedLevel.High`
2. send `setGoToSpeed(SpeedLevel.Slow)` to change moving speed to slow
3. Temi still moving to the destination with High speed

Test 2:
1. send `goTo()` command to Temi start from location A to B using `SpeedLevel.Slow`
2. send `setGoToSpeed(SpeedLevel.High)` to change moving speed to slow
3. Temi will use High speed to move to the destination

That means the Temi can only speed up, but cannot speed down?

temi SDK version: 0.10.81
launcher version: 16398-usa
robox version: 129.06
fireware version: 20220622.200917

zjn0505, 2022-08-16:
Thanks for pointing out.

`setGoToSpeed` while it is going doesn't work on 128 versinon.

We will add this support in 129 version.

kakeung, 2022-08-19:
Which new version should I waiting for? Is the Temi firmware version?

zjn0505, 2022-08-19:
It shall be in the 129 release.

Currently with 128 version, the workaround is stop and goto again with new speed level.

kakeung, 2022-08-22:
How to upgrade? My Temi's robox version is already version 129.06. 

On Temi 'Settings' -> 'About' -> 'Check for update', nothing new version for me.

zjn0505, 2022-08-22:
The 129 beta version doesn't have it yet.
The fix was made in last week and it wil be released in 129 publicly this week.

kltse, 2024-10-29:
my version is 134.07, my test case is to move temi from point A to B with goto(Speed) = slow, and then after 10 secs set high by setGoToSpeed(SpeedLevel.High). from my observation there is no changes in speed. while when on be with me mode, can obviously see the robot can move faster. please help.

zjn0505, 2024-11-07:
> my version is 134.07, my test case is to move temi from point A to B with goto(Speed) = slow, and then after 10 secs set high by setGoToSpeed(SpeedLevel.High). from my observation there is no changes in speed. while when on be with me mode, can obviously see the robot can move faster. please help.

Please make sure your app has applied SETTINGS permission to change the go to speed. It should work when changing the speed during movement.

You can try the same thing from temi center. Trigger a goto, then change the speed from temi center.

In 134 version, the goto speed settings doesn't affect "be with me" follow speed. It will be provided for configuration in 135 version.  https://github.com/robotemi/sdk/pull/476/files

kltse, 2024-11-07:
> Please make sure your app has applied SETTINGS permission to change the go to speed. It should work when changing the speed during movement.
>
> You can try the same thing from temi center. Trigger a [...]

thanks for your reply. can you provide screenshot on settings permission? thankyou

zjn0505, 2024-11-10:
> thanks for your reply. can you provide screenshot on settings permission? thankyou

It is in the first part of this method

https://github.com/robotemi/sdk/blob/master/sample/src/main/java/com/robotemi/sdk/sample/MainActivity.kt#L1614-L1635