robotemi/sdk #330: Disentangle manual and voice-activated wakeup toggle

ericwTR, 2022-09-20:
Hello,

I have a scenario where it is desirable to programmatically trigger Temi to wake up in order to start ASR/STT, but do not want Temi to ever actively listen for the wakeup phrase ("Hi Temi"). From my understanding enabling/disabling the wakeup setting, whether through the toggleWakeup() sdk call or through the Temi settings menu, applies to both the wakeup() sdk call and to voice activation.

I know that there are various conceivable ways to get around this (e.g. toggling the setting before/after calling wakeup(), handling whether to respond or not in the listener, etc.), but I was wondering whether all that I've said so far is correct, and if not how to do what I describe in the first sentence. It would be nice to be able to programmatically trigger wakeup such that the resulting Conversation Layer->ASR->NLU (or whatever subset of that) pipeline is executed without needing the agent to be constantly listening for the wakeup word or repeatedly turning it on and off.

Let me know if that all makes sense, thanks

zjn0505, 2022-09-21:
I think all of that makes sense, I will check with team on it.

If there would be a fix, it will only be appliable to temis running with new temi launcher in the coming release, as it would be a change from temi launcher, not temi SDK.

Currently the workaround will be as you said, calling the methods in a row.
```
toggleWakeup(false); wakeup(); toggleWakeip(true)
```

zjn0505, 2022-09-27:
We will change it as your suggestion to only block voice-actived wakeup in next software release, version 130.

closed 2022-09-27 by zjn0505